OrcaSlicer/OrcaSlicer #6628: Temperature tower non-manifold in Linux

FireflyJar, 2024-09-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.3 Cin

### Additional system information

Temperature tower non-manifold in Linux and Fix Model not available in Linux.

[image]

### Printer

any

### How to reproduce

Calibration > Temperuture

### Actual results

see screenshot

### Expected results

usable test stl

### Project file & Debug log uploads

[3mf.zip](https://github.com/user-attachments/files/16829145/3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

FireflyJar, 2024-09-01:
looks like it was fixed at one time
https://github.com/SoftFever/OrcaSlicer/issues/343
but still happening in Linux

Ergonomicmike, 2024-09-02:
https://github.com/SoftFever/OrcaSlicer/issues/3349

67RS427, 2024-10-28:
I'm seeing this bug as well on the latest v2.2.0 on MacOS

Lud00, 2024-11-02:
Same issue on Windows on 2.1.0 and 2.2.0
[image]

melbiz447, 2024-11-12:
Same issue here, but cannot fix the model on Mac. Had it on the older version too. Can we resolve this issue? Orca is way better than my primary slicer, but with a new filament...
[image]

67RS427, 2024-11-12:
Same issue here on MacOS with Orca v2.2... Please fix, thanks!

wisess, 2024-12-07:
temperature_tower.stl same issue on the last version

yanfali, 2024-12-16:
I'm also seeing this on 2.2.0 linux flatpak version

rali717, 2024-12-31:
Issue on  appImage version Ocrca Slicer 2.3.0 dev, too.

PaoloC68, 2025-02-19:
Since it is a pretty long time that this bug is around and since it appears that will not be fixed quickly I will recommend anyone that has this problem to go to this website https://www.formware.co/onlinestlrepair and fix the object. That is a far better temporary solution than have to install windows just to fix a 3d object.

hho, 2025-05-22:
Happens on 2.3.0 on Windows as well (though of course on Windows there actually _is_ a repair function)

ElonSatoshi, 2025-11-10:
[image]

What is it going to do, remove the issue if it's ignored long enough? Lmao

closed 2026-01-29 by ianalexis