OrcaSlicer/OrcaSlicer #6631: Orca Slicer freezes after starting

n00b0ss, 2024-09-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

NixOS 24.05

### Additional system information

Host: Latitude 7390 2-in-1
Kernel: Linux 6.10.5

 WM: Sway (Wayland)

CPU: Intel(R) Core(TM) i5-8350U (8) @ 3,60 GHz
GPU: Intel UHD Graphics 620 @ 1,10 GHz [Integrated]
Memory: 3,54 GiB / 7,63 GiB (46%)

### Printer

Voron V0.2

### How to reproduce

Start Orca-Slicer

### Actual results

Gets Stuck in Startscreen or if you are lucky you can create and try to open an STL File
Window is frozen and does not react.
After 30 s - 1 min Window closes from its own

Terminal Output:

orca-slicer:21591): Gtk-CRITICAL **: 20:00:11.703: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21591): Gtk-CRITICAL **: 20:00:11.703: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21591): Gtk-CRITICAL **: 20:00:11.710: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21591): Gtk-CRITICAL **: 20:00:11.713: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21591): Gtk-CRITICAL **: 20:00:11.713: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:21591): Gtk-CRITICAL **: 20:00:11.763: gtk_widget_set_size_request: assertion 'width >= -1' failed
[1]    21591 segmentation fault (core dumped)  orca-slicer

### Expected results

Slicer should work normally

### Project file & Debug log uploads

[warning]	2024-09-01 19:44:42.939818[Thread 0x00007f5866a24840]:Failed to load to hints.cereal. File does not exists. /home/basti/.config/OrcaSlicer/user/hints.cereal
[error]	2024-09-01 19:44:44.697654[Thread 0x00007f5866a24840]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-09-01 19:44:46.813286[Thread 0x00007f5866a24840]:calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-09-01 19:44:47.106805[Thread 0x00007f5866a24840]:LoadProfileFamily: parse /nix/store/cx3m4x0zxkrd7j2cg3axmvzspk8i0bpn-orca-slicer-2.1.1/share/OrcaSlicer/profiles/Dremel.json got exception: [json.exception.type_error.304] cannot use at() with string
[error]	2024-09-01 19:44:47.523975[Thread 0x00007f5866a24840]:LoadProfileFamily: parse /nix/store/cx3m4x0zxkrd7j2cg3axmvzspk8i0bpn-orca-slicer-2.1.1/share/OrcaSlicer/profiles/Kingroon.json got exception: [json.exception.type_error.302] type must be string, but is null
[error]	2024-09-01 19:44:47.896700[Thread 0x00007f5866a24840]:LoadProfileFamily: parse /nix/store/cx3m4x0zxkrd7j2cg3axmvzspk8i0bpn-orca-slicer-2.1.1/share/OrcaSlicer/profiles/Creality.json got exception: [json.exception.type_error.302] type must be string, but is null
[error]	2024-09-01 19:45:05.049103[Thread 0x00007f5866a24840]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-09-01 19:45:05.077092[Thread 0x00007f5866a24840]:get_version, get_version not supported,return 00.00.00.00!
[error]	2024-09-01 19:45:07.059354[Thread 0x00007f5866a24840]:calc_exclude_triangles:Unable to create exclude triangles

### Checklist of files to include

- [X] Log file
- [ ] Project file

UltraBlackLinux, 2024-09-16:
Same issue here
Something with fonts apparently - again?? But different?
```
Stack trace of thread 65006:
                #0  0x00005e585af62e74 tt_face_get_paint_layers (orca-slicer + 0x3be9e74)
                #1  0x00007132af6164de n/a (libcairo.so.2 + 0xa34de)
                #2  0x00007132af5c79ad n/a (libcairo.so.2 + 0x549ad)
                #3  0x00007132af5c7c83 cairo_scaled_font_glyph_extents (libcairo.so.2 + 0x54c83)
                #4  0x00007132afff8242 n/a (libpangocairo-1.0.so.0 + 0x8242)
                #5  0x00007132af6bf779 pango_glyph_string_extents_range (libpango-1.0.so.0 + 0x15779)
                #6  0x00007132af6d801a n/a (libpango-1.0.so.0 + 0x2e01a)
                #7  0x00007132af6d9253 n/a (libpango-1.0.so.0 + 0x2f253)
                #8  0x00007132af6cdb46 n/a (libpango-1.0.so.0 + 0x23b46)
                #9  0x00007132af6cdf6c n/a (libpango-1.0.so.0 + 0x23f6c)
                #10 0x00007132af6ce26e pango_layout_get_size (libpango-1.0.so.0 + 0x2426e)
```

n00b0ss, 2024-11-26:
Adding a port number to the Host helped.

(https://myside-yourside.net/@StarkRG/113526784879479099)

n00b0ss, 2024-11-26:
If you can't get to the setting, you might be able to add it manually by editing the printer's json file in .config/OrcaSlicer/user/default/machine/ the setting to change is "print_host

closed 2024-11-26 by n00b0ss