OrcaSlicer/OrcaSlicer #6634: Add Arc Overhangs to Orcaslicer

scruff2, 2024-09-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This is a very useful feature that has already been implemented in other slicers, such as Prusa Slicer.
Printers sometimes use a lot of overhang support material, especially when printing tall parts.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Reference:
https://github.com/SoftFever/OrcaSlicer/issues/4621
"It should incorporate Arc Overhangs, a technique described by Steven McCulloch."

boromyr, 2024-09-06:
+1 for this feature, the photo shows how the bridges would look like when built with Arc Overhangs:
[image]

made with this modified Arc Overhangs script, but it's not exceptional.
[parc.py.txt](https://github.com/user-attachments/files/16908910/parc.py.txt)

eugenz3rr, 2024-09-25:
Would also love to see this as a feature 🙏 even as a beta option

grrminator, 2024-11-20:
I am also interested in this as well @SoftFever

novarlynx, 2024-12-14:
+1 for this as well. Would love to see this implemented

TheOverpassArsonist, 2024-12-31:
As of #409 this feature was silently marked as "not planned" with no real explanation or reason that I can find. A decision I personally _strongly_ disagree with (for reasons I mention in 409) and I definitely think that decision should be reconsidered, but that is the current state of arc-overhangs nonetheless.

Anthony-Bec, 2025-01-03:
> +1 for this feature, the photo shows how the bridges would look like when built with Arc Overhangs: ![P_20240902_153452-01](https://private-user-images.githubusercontent.com/52782032/365188778-d1288f51-2735-4fe1-9bd9-1d5e0300447d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU4NzQwMjIsIm5iZiI6MTczNTg3MzcyMiwicGF0aCI6Ii81Mjc4MjAzMi8zNjUxODg3NzgtZDEyODhmNTEtMjczNS00ZmUxLTliZDktMWQ1ZTAzMDA0NDdkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwM1QwMzA4NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMzhkZDgzZjA2NDY5MjM5MWU4YTVmNDU0YWI1Zjk0MWUzYzE1NmUyNTVhOWVlOWZjNDE5ZDg1MGI2OTYzY2FmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.6XkPQKt7TYP31TdLNgm4QfS80qpEba4l8_we26glRnU)
> 
> made with this modified Arc Overhangs script, but it's not exceptional. [parc.py.txt](https://github.com/user-attachments/files/16908910/parc.py.txt)

how did you do this? its not working for me...

Pegaroo, 2025-01-03:
> > +1 for this feature, the photo shows how the bridges would look like when built with Arc Overhangs: ![P_20240902_153452-01](https://private-user-images.githubusercontent.com/52782032/365188778-d1288f51-2735-4fe1-9bd9-1d5e0300447d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU4NzQwMjIsIm5iZiI6MTczNTg3MzcyMiwicGF0aCI6Ii81Mjc4MjAzMi8zNjUxODg3NzgtZDEyODhmNTEtMjczNS00ZmUxLTliZDktMWQ1ZTAzMDA0NDdkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwM1QwMzA4NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMzhkZDgzZjA2NDY5MjM5MWU4YTVmNDU0YWI1Zjk0MWUzYzE1NmUyNTVhOWVlOWZjNDE5ZDg1MGI2OTYzY2FmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.6XkPQKt7TYP31TdLNgm4QfS80qpEba4l8_we26glRnU)
> > made with this modified Arc Overhangs script, but it's not exceptional. [parc.py.txt](https://github.com/user-attachments/files/16908910/parc.py.txt)
> 
> @boromyr how did you do this? its not working for me...

I haven't tried this myself but have a look at this GitHub 
https://github.com/Kelsch/arc-overhang-orcaslicer-integration

boromyr, 2025-01-10:
> > +1 per questa funzione, la foto mostra come apparirebbero i ponti se costruiti con Arc Overhangs:[ <img alt="P_20240902_153452-01" width="161.8125" height="21" src="https://private-user-images.githubusercontent.com/52782032/365188778-d1288f51-2735-4fe1-9bd9-1d5e0300447d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU4NzQwMjIsIm5iZiI6MTczNTg3MzcyMiwicGF0aCI6Ii81Mjc4MjAzMi8zNjUxODg3NzgtZDEyODhmNTEtMjczNS00ZmUxLTliZDktMWQ1ZTAzMDA0NDdkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwM1QwMzA4NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMzhkZDgzZjA2NDY5MjM5MWU4YTVmNDU0YWI1Zjk0MWUzYzE1NmUyNTVhOWVlOWZjNDE5ZDg1MGI2OTYzY2FmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.6XkPQKt7TYP31TdLNgm4QfS80qpEba4l8_we26glRnU"> ](https://private-user-images.githubusercontent.com/52782032/365188778-d1288f51-2735-4fe1-9bd9-1d5e0300447d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU4NzQwMjIsIm5iZiI6MTczNTg3MzcyMiwicGF0aCI6Ii81Mjc4MjAzMi8zNjUxODg3NzgtZDEyODhmNTEtMjczNS00ZmUxLTliZDktMWQ1ZTAzMDA0NDdkLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwM1QwMzA4NDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMzhkZDgzZjA2NDY5MjM5MWU4YTVmNDU0YWI1Zjk0MWUzYzE1NmUyNTVhOWVlOWZjNDE5ZDg1MGI2OTYzY2FmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.6XkPQKt7TYP31TdLNgm4QfS80qpEba4l8_we26glRnU)
> > realizzato con questo script Arc Overhangs modificato, ma non è eccezionale. [parc.py.txt](https://github.com/user-attachments/files/16908910/parc.py.txt)
> 
> @boromyrcome hai fatto? non funziona per me...

It's been a while since I used that script, what error does it give you?
The script has been recently updated here: https://github.com/Wasupmacuz/arc-overhang-prusaslicer-integration/tree/arc-overhangs-v1.0.0-beta but it doesn't support Orca yet, I will make it compatible with Orca as soon as I can.

Anthony-Bec, 2025-01-11:
@boromyr I get the error on BS and Orca Slicer
Here is the error: Failed starting the script C:\Users\anthony\Downloads\parc.py.txt
C:\Users\anthony\AppData\Local\Temp/bamboo_model/Fri_Jan_10/21_06_02#3620#55/Metadat
a\.3620.0.gcode, Win32 error: 193

boromyr, 2025-01-11:
You have to rename it to .py and then run it with python.exe.

boromyr, 2025-01-13:
Support for orcaslicer has been added.
https://github.com/Wasupmacuz/arc-overhang-prusaslicer-integration/blob/arc-overhangs-v1.0.0-beta/arc_overhangs_v1.0.0.py

pickanothername, 2025-01-22:
The Arc Overhang PrusaSlicer integration now also supports OrcaSlicer. This may make it easier to implement:

https://github.com/Wasupmacuz/arc-overhang-prusaslicer-integration/blob/arc-overhangs-v1.0.0-beta/arc_overhangs_v1.0.0.py

Jean-DrEaD, 2025-01-28:
Not working here, tested in 2.2.0 and 2.3.0-dev
I'm doing something wrong?

[image]

axiopaladin, 2025-03-11:
> Not working here, tested in 2.2.0 and 2.3.0-dev I'm doing something wrong?
> 
> [image]

You need to install the `shapely` libraray from pip, probably.

Maximus-CZ, 2025-04-09:
This needs to be added so bad. Prusa slicer has it, cant we just pull their implementation?

yves5141, 2025-04-10:
I would love to see this added, too.

jeefan3d, 2025-06-10:
This would be a great feature