OrcaSlicer/OrcaSlicer #6639: minimum layer time issue

Dconn88, 2024-09-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

qidi X Max 3

### How to reproduce

Use any STL that hits the minimum layer time for most of the layers and duplicate it so there are 2 of them on the build plate.

### Actual results

 total printing time is double compared to printing one object..
[image]
[image]
[image]
[image]
[image]
[image]

### Expected results

Should only be slightly more time than printing one unit because a majority of the layers are limited by minimum layer time.

### Project file & Debug log uploads

[Dolby Atmos height speaker stands.zip](https://github.com/user-attachments/files/16842338/Dolby.Atmos.height.speaker.stands.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

xokahag, 2024-10-19:
Switch the display to "Layer Time (log)" and you can clearly see that only the top section has layer times of 6s, which is the minimum allowed according to the filament settings:
[image]

When duplicating the object only the very top of the print object gets printed slightly faster:
[image]

Most modern printers are limited by the maximum volumetric flow rate. Your filament has a max setting of 32mm³/s, which you pretty much archive when printing at 300mm/s, with your line width and layer height:
[image]

This issue can be closed.

Dconn88, 2024-10-20:
@avogra @xokahag 
Thank you guys, I understand what you're both saying. I completely agree.
I think the total print time is correct, and it's just very deceiving from the layer time colors from both the points you all brought up. 
 It totally threw me off and hindsight I felt a little foolish about putting this post, however it does bring up a good point...the colors and time intervals are really the problem here. It would be nice if We could select what range of layer times we would like to color code. For instance it could give the whole rainbow of colors to only the layers below 30 seconds. That way each color would have a very small range of a couple seconds.
Usually the only thing I'm looking at is short layer times to figure out cooling And also what my speed bottleneck is. We're trying to troubleshoot something about the print. 

Thanks so much for your time guys. Let me know what you think about this idea. Most of the work and calculation is already done, and this tweak would allow the layer time color coding to be so much more useful!

closed 2025-01-27 by github-actions[bot]