OrcaSlicer/OrcaSlicer #6642: Unsupported bridges

Darwinian999, 2024-09-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab X1 Carbon 0.4 nozzle

### How to reproduce

Load the attached 3MF file and slice it.

### Actual results

Bridges that are not walls have lines that start and end with no walls below them to attach to. With nothing to attach to, the bridges fail. Photo and 3MF file (in ZIP) attached.
[image]

### Expected results

The same 3MF file sliced in Bambu Studio has lines that start and end with walls below them to attach to, which is the expected behaviour. Photo attached.
[image]

### Project file & Debug log uploads

[Bridge Test2 - BS.zip](https://github.com/user-attachments/files/16847429/Bridge.Test2.-.BS.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2024-09-04:
I can confirm this on 2.1.1 as well. Will take a look soon.

SoftFever, 2024-09-04:
The difference between OS and BS is most likely caused by using different wall generator.
Can you try classic wall generator in Orca?  Or try Arachne in BS to verify the cause.

There are some edge cases that Arachne will not bridge well.

Darwinian999, 2024-09-04:
OS: The results are the same with Classic or Arachne walls
BS: The results are the same with Classic or Arachne walls

i.e. the wall type doesn’t change the result in either BS or OS.

Regards,
Chris

SoftFever, 2024-09-04:
Ah, now I recall it. 

Yes, this issue only occurs when there is only one bridge layer, which means the bridge margin isn't applied as it incorrectly considers it as a top layer.

@Noisyfox, you might find this information helpful when looking into this issue.

SoftFever, 2024-09-04:
@Darwinian999 
You may want to consider making the top surface at least 2-layer thick for now to carry on the test. 
Although I will say it's a good idea to have multiple layers on top anyway when benchmarking bridge performance as that give us better idea how well it supports the upper layers
.

Noisyfox, 2024-09-05:
> Ah, now I recall it.
> 
> Yes, this issue only occurs when there is only one bridge layer, which means the bridge margin isn't applied as it incorrectly considers it as a top layer.
> 
> @Noisyfox, you might find this information helpful when looking into this issue.

Cool, I've identified the issue. Will raise a PR soon.

Darwinian999, 2024-09-05:
> Although I will say it's a good idea to have multiple layers on top anyway when benchmarking bridge performance as that give us better idea how well it supports the upper layers

Yep, understood. I made this test to see if I could find optimal settings for the 1st layer for a bridge and then I'd move onto seeing if different settings helped for the next layer of the bridge, hoping to get the cleanest bridges possible (mainly for PETG, PA and PC). I'm probably over-thinking it though :)

Noisyfox, 2024-09-05:
> Yep, understood. I made this test to see if I could find optimal settings for the 1st layer for a bridge and then I'd move onto seeing if different settings helped for the next layer of the bridge, [...]

Nevertheless, your tests exposed a bug that needs to be fixed, which is valuable and is making the software better.

PR #6654 (2024-09-07): Fix bridge anchor when bridge is also the top surface
Fixes #6642

Cherry picked from https://github.com/prusa3d/PrusaSlicer/commit/b133579126e6cee44f841f66dbae018455e32117
Huge thanks for their excellent work!

[image]
commit 4e252179 (2024-09-07): Fix bridge anchor when bridge is also the top surface (#6654)

Fixes #6642

Cherry picked from https://github.com/prusa3d/PrusaSlicer/commit/b133579126e6cee44f841f66dbae018455e32117
Huge thanks for their excellent work!

[image]
closed 2024-09-07 by SoftFever