OrcaSlicer/OrcaSlicer #6646: Layer sliders in Preview section lose onclick focus when the cursor leaves the app window

einichi, 2024-09-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

14.6.1

### Printer

Ankermake M5

### How to reproduce

1. Slice any file and go to Preview tab
2. Click, hold and drag the layer slider up and down, note it should work normally
3. Now click and drag, but move the mouse outside of the Orca Slicer window
4. The slider should lose its click focus and the camera of the 3D view will move instead

### Actual results

Preview camera moves instead of the layer slider

### Expected results

Layer slider continues to move even if cursor drifts outside of app window

### Project file & Debug log uploads

Project is irrelevant to issue

[debug_Tue_Sep_03_01_00_02_3949.log](https://github.com/user-attachments/files/16860970/debug_Tue_Sep_03_01_00_02_3949.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

https://github.com/user-attachments/assets/d5bedb0a-7cea-477f-b9cb-f772b30fada3

Attaching short screen capture of the issue

Rekemot, 2024-10-04:
I join the question. It's very frustrating when you want to check something and all of a sudden the whole preview starts rotating instead of changing layers just because the mouse went outside the window.

einichi, 2024-10-04:
I should add that I do not experience this behavior in any other slicer, including derivatives.

LazaroFilm, 2024-10-22:
Same problem here. It's especially hard on my setup with dual monitors as my cursor goes to the right monitor and the view get's all messed up. 

https://github.com/user-attachments/assets/bc125013-a6f3-45b5-a073-4cae78239ae0

einichi, 2024-12-11:
Issue also opened at https://github.com/bambulab/BambuStudio/issues/5515

closed 2025-03-19 by github-actions[bot]