OrcaSlicer/OrcaSlicer #6647: Prime all extruders Tools are not respecting prime volume settings

traxman25, 2024-09-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 Dev released Sept 4

### Operating System (OS)

Windows

### OS Version

10

### Printer

StealthChanger tool changer

### How to reproduce

After saving printer, filament and process settings for a multi tool printer then reslicing.

### Actual results

The prime lines at the beginning of the print overall are not respecting Prime Volume setting in Process > Multimaterial > Prime Tower > Prime Volume. Additionally the last tool to prime (tool 0 in my case) purges a much higher volume than the rest of the tools. 
[image]

### Expected results

All tools prime lines should use the same prime volume and use the setting in Process > Multimaterial > Prime Tower > Prime Volume for the volume purged.

### Project file & Debug log uploads

[prime line bug report.zip](https://github.com/user-attachments/files/16861550/prime.line.bug.report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

AstronIsTaken, 2024-09-06:
Having the same with my SC as well. Last tool's purge has much higher volume
[image]

Ataraxia-Mechanica, 2024-09-09:
Apparently this can be fixed by turning on single extruder multi material, set all flushing volume to 0, then disable it.

traxman25, 2024-09-09:
That in itself is also a bug however, I've already done that. That only fixes the prime tower. Flushing volume does not fix the start of print prime lines at the edge of the build plate, which is what this bug report is about.

traxman25, 2024-09-09:
Cross posting from my reply to 6699: 
And it just went away.... Changed to single, changed flushing multiplier to 0.1, changed back to multi tool and got the small prime line. Changed back to single, changed flushing multiplier to 0, changed back to multi tool. Slice and it had a large prime line again. When going back to multi tool each tool had changed the 'Extra length on start' to 0.2 so I changed that back to 0.1 like I had it before, sliced and the extra prime is gone. Saved the profile and the extra prime is gone.

I'm guessing there's too many "extra prime/purge/flush volume" settings in too many different places and in too many places that can only be adjusted in certain configurations which is causing the program to get mixed up.

ningpj, 2024-10-29:
The other related issue is being able to disable the `Prime tower: Prime all printing extruders` option when you want to. I can search for the option in 2.2.0 GA but it doesn't appear to have been map to a UI option under Process > Multi Material or else where in the UI to disable it.  

I am also seeing the same issue and behaviour and prime line issue in 2.2 GA.

closed 2025-02-05 by github-actions[bot]