OrcaSlicer/OrcaSlicer #6648: Skirt loops variable is not respected after 1st layer with multi material multi tool printing

traxman25, 2024-09-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 Dev released Sept 4

### Operating System (OS)

Windows

### OS Version

10

### Printer

StealthChanger tool changer

### How to reproduce

Slice any multi tool project with more than (1) set for both Skirt Loops and Skirt Height variables.

### Actual results

The number of Skirt Loops is correct for the first layer. If using a skirt for more than one layer (2 in this case) then the number of loops is ignored, and only one loops is performed. Changing the prime volume setting in either 'Filament > Multimaterial > Wipe tower parameters > Minimum purge on wipe tower' or 'Process > Multimaterial > Prime Tower > Prime Volume' does not affect the results of the skirt settings with or without either of the flushing options in the section 'Process > Multimaterial > Flushing Options' options selected or not selected. 
[image]

### Expected results

I make the skirts 2 layers thick to make them easier to remove from the build plate. This requires the skirt to be printed with the same number of loops for all layers. When setting skirt settings to 4 loops, and 2 layers on a single color/tool printer both layers are 4 loops.

### Project file & Debug log uploads

[prime line bug report.zip](https://github.com/user-attachments/files/16861710/prime.line.bug.report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kybarg, 2024-09-09:
is it related to #6694

traxman25, 2024-09-09:
I'm only doing skirt loops but it does look like the same root issue.
That said I have noticed that this issue does occur on all of my printers/profile, not just multitool.

SpeCter, 2024-10-12:
Yeah this bug is kinda detrimental for materials that really don't like to leave the build plate

closed 2025-01-19 by github-actions[bot]