OrcaSlicer/OrcaSlicer #6651: Travel distance optimization between objects

Scootworks, 2024-09-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Lets assume we have two objects, Obj1 and Obj2. The current logic prints in this order:

1. layer 1 on Obj1
2. layer 1 on Obj2
3. layer change
4. layer 2 on Obj1
5. layer 2 on Obj2

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

To decrease travel distance and total print time, it would make sense to print in this order:

1. layer 1 on Obj1
2. layer 1 on Obj2
3. layer change
4. layer 2 on Obj2
5. layer 2 on Obj1

to further improve this idea, the logic for the "next" object could be calculated be the distance from the current object position, in case you have more than 2 objects. the closest start point of the next object should be the next printed object.

vgdh, 2024-09-04:
As far as I know that had been discussed and be concluded that it will make worse surface finish due uneven cooling.

vgdh, 2024-09-04:
But I've not seen any evidence or tests that prove that conclusion.

Scootworks, 2024-09-04:
that can, of course be an issue, but for many parts, that wouldn‘t be an issue at all

vgdh, 2024-09-04:
I agree, may be some check-box for enable this order of print will be great!

Scootworks, 2024-09-04:
yeah for sure. I forgot to mention that

vgdh, 2024-09-04:
linking: https://github.com/SoftFever/OrcaSlicer/discussions/5132

Gudzonskiy, 2024-09-06:
I support the idea! So that there was a flag, whoever needs it will enable this option!

closed 2024-12-13 by github-actions[bot]