OrcaSlicer/OrcaSlicer #6658: Can't create a nozzle for an existing printer

lucasmontec, 2024-09-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

amd 5800x, gtx 3080ti

### Printer

Ender-3 V2

### How to reproduce

1. Open orca
2. Click to select a printer
3. Click to create a printer
4. Select to create a nozzle for an existing printer
5. Try to select the printer

### Actual results

Nothing happens, no printers show up.

### Expected results

The dropdown selection should have shown my configured 0.4mm nozzle ender-3 v2.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16898065/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lucasmontec, 2024-09-05:
https://github.com/user-attachments/assets/5cea57a7-5fba-4f21-9d2c-83ff8de60da3

Bat-Van, 2024-10-03:
I'm having the same issue

zhangxiaoxing, 2024-11-02:
I also have this issue

lucasmontec, 2025-01-23:
I can confirm the same issue happens on 2.2.0.

closed 2025-05-01 by github-actions[bot]

lucasmontec, 2025-07-05:
Still happening