OrcaSlicer/OrcaSlicer #6662: Option to force overhangs to be printed inner/outer wall order

CartesianFish, 2024-09-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When printing 90 degree overhangs with any wall order other than inner/outer, one or more wall will print in mid-air. This creates defects when printing counterbore holes in a model.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A boolean option to force all overhangs of a specifiable degree angle and greater from vertical, to ignore printing order and print last. Overhangs on outermost walls will be treated as inner/outer. Overhangs that are inside the boundaries of the part, AKA walls that fall inside the same group as "inner brim only" walls, would be printed inner/outer. All other layers, AKA layers without applicable overhangs, will be printed using the set wall order. The option to paint specific overhangs and set their order in relativity to the standard wall order would also be nice.

### Describe alternatives you've considered

If arc overhangs are implemented then this might be part of making arc overhangs more applicable to models that require any wall order other than inner/outer.

### Additional context

[image]
[image]
The model shown can't use inner/outer order, so shown are both inner/outer/inner and outer/inner wall orders to explain the overhang type referenced before.

vgdh, 2024-09-06:
that would be great to implement, I also often encounter these situations with overhangs.

igiannakas, 2024-09-07:
It will cause visible artefacts on the walls surrounding the overhang. Hence it’s not implemented. The wall ordering must remain the same throughout the print to avoid builders from switching to inner outer (as the inner wall pushes out the outer wall).

CartesianFish, 2024-09-07:
> It will cause visible artefacts on the walls surrounding the overhang.

Couldn't it be useful for non-visible walls, like in the counterbore hole example? It doesn't need to be pretty, just prevent printing in mid-air for that hole. Would it be better to make it a modifier? So the user can use mesh modifiers in specic places where the walls won't be seen.

igiannakas, 2024-09-07:
> Couldn't it be useful for non-visible walls, like in the counterbore hole example? It doesn't need to be pretty, just prevent printing in mid-air for that hole. Would it be better to make it a [...]

In a counterbore hole you can use the sacrificial bridging option and you won’t be printing in mid air at all ;)

CartesianFish, 2024-09-07:
The counterbore hole example was simply the best I could come up with at 3am where I am. There are non-hole related model features that won't always be seen but can't use sacrificial bridging, like print in place models. The ability to simply share the .3mf file with the mesh modifiers could expand possibilities.

vgdh, 2024-09-07:
> It will cause visible artefacts on the walls surrounding the overhang. Hence it’s not implemented. The wall ordering must remain the same throughout the print to avoid builders from switching to inner outer (as the inner wall pushes out the outer wall).
> 
> implementing an option like this defeats the purpose of using inner outer inner or outer inner which is smoother walls.

If the printed wall is sagging it certainly ruin the smooth surface. 😉
May be increase line width on such overhangs partially solve this problem, but it needs testing.

slothy89, 2024-09-07:
> The counterbore hole example was simply the best I could come up with at 3am where I am. There are non-hole related model features that won't always be seen but can't use sacrificial bridging, like [...]

A simple example to aide your point would be the CaliDragon where it starts to print the chin. This is an external overhang so counterbore bridging is irrelevant. The chin definitely prints cleaner with an inner-outer order.

[image]

Rickthebig, 2024-10-05:
I very much support this enhancement, as the inconsistency it alledgely introduces (who has tested that?) would be still better than a misscarried overhang.

Ryan-gsq, 2024-11-05:
it's very useful

GKSN, 2024-11-28:
It would be very useful, I'm having a lot problems just because of this. I would help mainly on non-visible walls ,the counterbore hole for example...

closed 2025-03-06 by github-actions[bot]

TooMuchAndNotEnough, 2025-08-02:
I know this issue is closed, but I'm running into the same problem in 2025. Would be nice to have a "force inner/outer order for overhangs above x%"

[image]

igiannakas, 2025-08-02:
This will look worse overall on the model. The best solution here is either supports and inner outer inner or print inner outer throughout.

Swapping wall order mid print introduces bulges across all features of that layer which will look no better than a sagging overhang. So you’re swapping one artefact from another.

If you want to observe these artefacts try enabling the reverse on even option. It will show you the type of refract that will be shown there.