OrcaSlicer/OrcaSlicer #6666: No prepare o preview redered

Idire, 2024-09-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.3

### Additional system information

### System:
  Kernel: 5.15.0-91-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
### Machine:
  Type: Desktop System: Apple product: iMac13,1 v: 1.0 
### CPU:
  Info: quad core model: Intel Core i5-3330S bits: 64 type: MCP 
### Graphics:
  Device-1: NVIDIA GK107M [GeForce GT 640M Mac Edition] vendor: Apple driver: nvidia v: 390.157
    pcie: speed: 2.5 GT/s lanes: 16 bus-ID: 01:00.0 chip-ID: 10de:0fd8 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 101 s-size: 483x272mm (19.0x10.7") s-diag: 554mm (21.8")
  Monitor-1: DP-2 res: 1920x1080 hz: 60 dpi: 103 size: 475x267mm (18.7x10.5")
    diag: 545mm (21.5")
  OpenGL: renderer: GeForce GT 640M/PCIe/SSE2 v: 4.6.0 NVIDIA 390.157 direct render: Yes

### Printer

RatRig V-core 5 500 0.6 nozzle

### How to reproduce

Just open the AppImage provided from releases

### Actual results

After opening the app a flat white screen is shown instead of the 3D space of the buildplate

[image]

### Expected results

The preview and prepare tabs should render a 3d environment.

### Project file & Debug log uploads

[debug_Fri_Sep_06_22_38_31_71299.log.0.zip](https://github.com/user-attachments/files/16912787/debug_Fri_Sep_06_22_38_31_71299.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2024-09-07:
Could you try PR https://github.com/SoftFever/OrcaSlicer/pull/6621 and see if that changes anything?

Idire, 2024-09-08:
> Could you try PR #6621 and see if that changes anything?

I haven't been able to compile OrcaSlicer, my machine just end up hanging after a few hours, so I have not been able to run that pull request to see if that would solve the problem.

istinet, 2024-09-08:
I have the same problem on Arch after updating. 
OrcaSlicer Version: 2.1.1
Operating System: EndeavourOS 
KDE Plasma Version: 6.1.4
KDE Frameworks Version: 6.5.0
Qt Version: 6.7.2
Kernel Version: 6.10.8-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5800X3D 8-Core Processor
Memory: 15.5 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 4070/PCIe/SSE2
Manufacturer: Micro-Star International Co., Ltd.
Product Name: MS-7C37
System Version: 3.0
NVIDIA Driver Version: 560.35.03

Noisyfox, 2024-09-08:
> > Could you try PR #6621 and see if that changes anything?
> 
> I haven't been able to compile OrcaSlicer, my machine just end up hanging after a few hours, so I have not been able to run that pull request to see if that would solve the problem.

You can directly download the CI build from here https://github.com/SoftFever/OrcaSlicer/pull/6621/checks

istinet, 2024-09-09:
I've found [this](https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-2336867684) post that resolved the issue for me. 

Tested on Arch with Nvidia 560.35.03 on KDE Wayland

__GLX_VENDOR_LIBRARY_NAME=mesa
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json

spusuf, 2024-09-10:
> I've found [this](https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-2336867684) post that resolved the issue for me.
> 
> Tested on Arch with Nvidia 560.35.03 on KDE Wayland
> 
> __GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json

Just wanted to add context from my post because it's missing here: This enables software rendering which is not ideal, you've got a GPU for a reason. But it's the workaround for NVIDIA cards/Wayland/bambuslicer for the time being.

Idire, 2024-09-11:
> I've found [this](https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-2336867684) post that resolved the issue for me.
> 
> Tested on Arch with Nvidia 560.35.03 on KDE Wayland
> 
> __GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json

That has fixed the problem for now, but quite heavy on the CPU. Hope it gets fixed upstream soon.

harry42203, 2024-10-19:
> I've found [this](https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-2336867684) post that resolved the issue for me.
> 
> Tested on Arch with Nvidia 560.35.03 on KDE Wayland
> 
> __GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json

Newbie question, but how do I set these ENV variables?

spusuf, 2024-10-20:
> Newbie question, but how do I set these ENV variables?

You can launch from command line add those in the same line before launching your program. Or if you use something like Gear Lever (would recommend) for AppImages you can click add to menu then add an environment variable at the bottom.

TheDevMinerTV, 2024-11-03:
> I've found [this](https://github.com/SoftFever/OrcaSlicer/issues/6433#issuecomment-2336867684) post that resolved the issue for me.
> 
> Tested on Arch with Nvidia 560.35.03 on KDE Wayland
> 
> __GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json

On Arch Linux you have to check that you have `mesa` installed instead of `mesa-amber` (which is an older version of Mesa). Uninstalling `mesa-amber` can make your displays not work properly in GPU accelerated things (like drawing your desktop).

closed 2025-02-09 by github-actions[bot]