OrcaSlicer/OrcaSlicer #6671: Preview - adjustable model opacity/transparency

Nebur99, 2024-09-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In the Preview the design model is barely discernible with minimal opacity.
When trying to optimize things, comparing the design model with the sliced output is very helpful. Right now it's not really possible because it's barely visible.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add two sliders in the **Preview** that adjust  **sliced model opacity** and **design model opacity**.
Positioning the sliders in the upper area of the view would work nicely from a users point of view.

### Describe alternatives you've considered

Alternatively, with option for future enhancements:
- add an extra "Color scheme" called "Comparison" (like Line Type, Line Width, Flow etc.) 
- _for now:_  just contains the opacity sliders described and color pickers to choose the two colors to use for design and sliced model
- _in the future:_ compares design and sliced model and indicates deviations outside of a user defined range (max plus/minus mm) by a color scale (this a typical feature of CAM software in the machining world)

philsward, 2024-09-09:
100% would like this added for the "Prepare" page. Ideally, select a model -> right click -> opacity

The "workaround" that people have pointed out is to add a "fill" using multiple filaments. This is unnecessary and not an ideal workflow when the model needs to have transparency for adding modifiers, support blockers, etc.

The only option for modifiers on models that are enclosed is to guess where to put them, slice, move, slice, wash-rinse-repeat.

I haven't had a need for changing the opacity on the preview page as suggested by the OP but on the Prepare page? Definitely needed.

closed 2024-12-17 by github-actions[bot]

CombatVolcano, 2025-01-14:
this is so annoying. why does it have this semi transparent model view. I need to be able to see the colors on the print, not the colors on the stupid preview

mxxmxxm2, 2026-10-15:
Having a transparent view is essential when setting up a 3D model for printing. Right now, it only appears for a moment during the slicing process. Could we get a dedicated button or menu option to turn this view on without needing to slice the model first?