OrcaSlicer/OrcaSlicer #6672: Pressure Advance Pattern Test Using Wrong Layer Height on Different Nozzle Sizes

OsmosisD, 2024-09-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU: AMD Ryzen 7 5800X
RAM: 64 Gb
GPU: NVIDIA GeForce GTX 1650 SUPER

### Printer

Bambu Lab P1S

### How to reproduce

1. Use a 0.2mm nozzle (or other non-0.4mm size)
2. Click Calibration up the top.
3. Select "Pressure Advance".
4. Select "PA Pattern" and hit OK.

### Actual results

The automated test generation tries to give us a layer height of 0.2mm (with 0.25mm for the initial layer), resulting in a slicing error because the layer height can't exceed the nozzle diameter.

[image]

I tried editing the layer heights manually and reslicing, which correctly changes the layer height for the custom PA pattern lines, but after printing there's no discernible difference between a PA of 0 and a PA of 0.12 on PLA, which is bizarre, like it's not adjusting PA at all. And yes, with this nozzle size I do need a magnifying glass to tell, lol

### Expected results

It should adjust the layer height according to the nozzle diameter. It does fine for the line width.

Also not sure if changing the layer height manually messes up the PA test. That might be a separate bug.

### Project file & Debug log uploads

[Broken Pressure Advance Test.zip](https://github.com/user-attachments/files/16919851/Broken.Pressure.Advance.Test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Using a Revo 0.25mm ObXidian nozzle on a Panda Revo, if relevant. The issue happens with a stock 0.2mm nozzle too, however.

Ergonomicmike, 2024-09-07:
> like it's not adjusting PA at all. 

Does the generated gcode file show M117 PA changes or SET_PRESSURE_ADVANCE ?

```
; start pressure advance pattern for layer
G1 E-2.3 F2700 ; retract
G1 X65.461 Y86.007 F7500 ; Move to start XY position
G1 E2.3 F2520 ;  ; unretract
G1 Z.191 F7500 ; Move to start Z position
SET_PRESSURE_ADVANCE ADVANCE=0.4; Override pressure advance value
G1 E-2.3 F2700 ; retract
G1 X65.461 Y86.007 F7500 ; Move to box start
G1 E2.3 F2520 ;  ; unretract
G1 F1200
G1 X65.461 Y128.433 E2.12121 ; Draw perimeter (up)
```

OsmosisD, 2024-09-08:
I'm on a Bambu printer, so it's using the M900 command to set it, but it does appear to be doing that, yes.
I did some test prints later and pressure advance is working correctly on a 0.25mm nozzle, but the effects are much harder to recognize than with a 0.4mm nozzle. So seems to be something particular to very low volumetric flow rates.
I haven't tested a wide range of values yet but there was no noticeable effect from PA at 0.12. At a level of 1 (for a 1 second delay) there were significant extrusion errors in the model, as expected, though not nearly as bad as they'd have been if printed using a 0.4mm nozzle.

Ergonomicmike, 2024-09-08:
Either due to very low volumetric flow rates or perhaps the high resistance caused by a small nozzle. Which might act to dampen out changes in flow rate. 

Or both.

(My intuition/WAG.)

OsmosisD, 2024-09-08:
I am using glow-in-the-dark filament, which I've noticed I have to bump up flow rate a lot for the smaller nozzle (no such problem with regular PLA), probably due to the filler impeding flow. Anyway, that issue aside, the layer height is still incorrect by default.

closed 2024-12-15 by github-actions[bot]