OrcaSlicer/OrcaSlicer #6684: Export prime tower as an excludable objects so Klipper adaptive mesh will include the prime tower.

IRTrail, 2024-09-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When Klipper does an adaptive mesh, it does not include the prime tower, which I usually place well away from the print because I use no sparse layers. This causes the mesh to not include the prime tower. The result is either dragging nozzles or not adhering to the bed at the prime tower.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

It would be nice if the prime tower is, or optionally, included in the exclude objects output.

### Describe alternatives you've considered

1. Exclude objects could be turned off and Klipper would mesh the entire bed, but then the ability to exclude objects during a print is lost.
2. Booze. They seem to fix everything else in life. Maybe that's what I need...more booze...?

Dylwad, 2024-09-12:
Add a small disc a few layers high on the opposite side of the prime tower from the model and it will be an object that forces the mesh to cover the prime tower.

IRTrail, 2024-09-12:
@Dylwad I have a working python script that makes the object in the code. It took all of an hour with my rather feebile python skills.  But, I am having trouble figuring out how to make it accept the input and return output if used through Orca's post processing script tool.

Thanks for the suggestion. I'll use it if I can't get the script working through Orca.

fritzw, 2024-10-15:
@IRTrail The input file name is in `sys.argv[1]` in Python. The output must be written to the same file. So either you write the output to a temporary file and then rename (overwrite) that to the input file, or you read the entire input file into a string, process it from there, and immediately write to the input file.

IRTrail, 2024-10-15:
> @IRTrail The input file name is in `sys.argv[1]` in Python. The output must be written to the same file. So either you write the output to a temporary file and then rename (overwrite) that to the [...]

I have a working solution now. Look here:

https://github.com/IRTrail/Orca-Slicer-Post-Processing-Scripts

It's unfortunate this isn't included in Orca Slicer natively. Hopefully it will be soon!

closed 2025-01-21 by github-actions[bot]