OrcaSlicer/OrcaSlicer #6688: Object skirt problem

GlauTechCo, 2024-09-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

macOS

### OS Version

14.2.1 Sonoma

### Additional system information

Macbook Air M1

### Printer

FlashForge M5p

### How to reproduce

Create two objects close together

### Actual results

Skirts should not overlap.
It should give a warning about overlapping.

### Expected results

Skirts overlap.

### Project file & Debug log uploads

[image]
[image]
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

foreachthing, 2024-09-09:
This has been discussed many times before. Over on [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/issues/316) too.
You'd have to "assemble" your parts first. 
Or, don't put them so close together or change skirt distance 🙄

GlauTechCo, 2024-09-09:
> This has been discussed many times before. Over on [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/issues/316) too. You'd have to "assemble" your parts first. Or, don't put them so close together or change skirt distance 🙄

The operation may be considered correct, but for amateur users, a pop-up warning should appear to make the conflict visible.

closed 2024-12-17 by github-actions[bot]