OrcaSlicer/OrcaSlicer #6693: Fails to build on the Arch Linux

FFY00, 2024-09-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

main

### Operating System (OS)

Linux

### OS Version

Arch Linux (as of 9 September 2024)

### Printer

NA

### How to reproduce

Try to build `main` from source on an up-to-date Arch Linux system.

I am using the following command:
```
cmake -Bbuild -GNinja -Wno-dev -DOPENVDB_FIND_MODULE_PATH=/usr/lib/cmake/OpenVDB/FindOpenVDB.cmake -DSLIC3R_GTK=3
```

### Actual results

The build fails due to a couple issues:
- API incompatibility with Boost >=1.85
- Incompatibility with OpenCV 4.10 (I didn't bisect the exact OpenCV version it started failing)
	- The headers were missing in the `libslic3r ` target, and there were unresolved links `opencv_imgproc` symbols on the `OrcaSlicer` target
- Unresolved link `webkit` symbols on the `OrcaSlicer` target (likely due to wxWidgets upstream changes)
- Unresolved link `boost_log_setup` symbols on the `liblibslic3r_cgal` target

### Expected results

The project should successfully build.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I have opened a couple PRs to fix these issues:
- https://github.com/SoftFever/OrcaSlicer/pull/6686
- https://github.com/SoftFever/OrcaSlicer/pull/6690
- https://github.com/SoftFever/OrcaSlicer/pull/6691
- https://github.com/SoftFever/OrcaSlicer/pull/6692

With all PRs applied, I can successfully build OrcaSlicer on Arch Linux.

That said, with the system wxWidgets, it is segfaulting. I'll try to debug that.

CitrusIntellect, 2024-09-18:
you can try using yay -S orca-slicer-bin,
I did and it worked great.

If you want to build it though you can use yay -S orca-slicer

FFY00, 2024-09-20:
On an up-to-date system, the pre-built binaries are seg-faulting.

```
Thread 1 "orcaslicer_main" received signal SIGSEGV, Segmentation fault.
0x0000555559181fb4 in tt_face_get_paint_layers ()
(gdb) bt
#0  0x0000555559181fb4 in tt_face_get_paint_layers ()
#1  0x00007ffff716c4de in _cairo_ft_scaled_glyph_is_colr_v1 (scaled_font=<optimized out>, scaled_glyph=0x55555bf24160, face=0x55555bf32800) at ../cairo/src/cairo-ft-font.c:3194
#2  _cairo_ft_scaled_glyph_init_metrics (scaled_font=<optimized out>, scaled_glyph=0x55555bf24160, face=0x55555bf32800, vertical_layout=0, load_flags=512, foreground_color=0x7ffff71d9c40 <cairo_color_black.lto_priv>) at ../cairo/src/cairo-ft-font.c:3268
#3  _cairo_ft_scaled_glyph_init (abstract_font=<optimized out>, scaled_glyph=0x55555bf24160, info=CAIRO_SCALED_GLYPH_INFO_METRICS, foreground_color=0x7ffff71d9c40 <cairo_color_black.lto_priv>) at ../cairo/src/cairo-ft-font.c:3381
#4  0x00007ffff711d9ad in _cairo_scaled_glyph_lookup (scaled_font=scaled_font@entry=0x55555bf36bd0, index=<optimized out>, info=info@entry=CAIRO_SCALED_GLYPH_INFO_METRICS, foreground_color=0x7ffff71d9c40 <cairo_color_black.lto_priv>, foreground_color@entry=0x0,
    scaled_glyph_ret=scaled_glyph_ret@entry=0x7fffffff27a0) at ../cairo/src/cairo-scaled-font.c:2913
#5  0x00007ffff711dc83 in cairo_scaled_font_glyph_extents (scaled_font=0x55555bf36bd0, glyphs=glyphs@entry=0x7fffffff27f0, num_glyphs=num_glyphs@entry=1, extents=extents@entry=0x7fffffff2810) at ../cairo/src/cairo-scaled-font.c:1643
#6  0x00007ffff7e97242 in compute_glyph_extents (cf_priv=0x55555becc3d8, glyph=441, entry=0x55555bf23a78) at ../pango/pango/pangocairo-font.c:807
#7  _pango_cairo_font_private_get_glyph_extents_cache_entry (cf_priv=0x55555becc3d8, glyph=441) at ../pango/pango/pangocairo-font.c:833
#8  _pango_cairo_font_private_get_glyph_extents (cf_priv=0x55555becc3d8, glyph=441, ink_rect=0x7fffffff28c0, logical_rect=0x7fffffff28d0) at ../pango/pango/pangocairo-font.c:870
#9  pango_cairo_fc_font_get_glyph_extents (font=0x55555becc360, glyph=441, ink_rect=0x7fffffff28c0, logical_rect=0x7fffffff28d0) at ../pango/pango/pangocairo-fcfont.c:130
#10 0x00007ffff79d0779 in pango_glyph_string_extents_range (glyphs=0x55555bf345f0, start=0, end=43, font=0x55555becc360, ink_rect=<optimized out>, logical_rect=0x7fffffff2980) at ../pango/pango/glyphstring.c:210
#11 pango_glyph_string_extents_range (glyphs=0x55555bf345f0, start=0, end=43, font=0x55555becc360, ink_rect=<optimized out>, logical_rect=0x7fffffff2980) at ../pango/pango/glyphstring.c:164
#12 0x00007ffff79e901a in pango_layout_run_get_extents_and_height (run=0x55555bf3d3e0, run_ink=run_ink@entry=0x7fffffff2a60, run_logical=0x7fffffff2980, run_logical@entry=0x0, line_logical=line_logical@entry=0x7fffffff2a50, height=height@entry=0x7fffffff2a4c)
    at ../pango/pango/pango-layout.c:5640
#13 0x00007ffff79ea253 in pango_layout_line_get_extents_and_height.part.0.lto_priv.0 (line=0x55555bf3bbc0, ink_rect=<optimized out>, logical_rect=<optimized out>, height=<optimized out>) at ../pango/pango/pango-layout.c:5836
#14 0x00007ffff79deb46 in pango_layout_line_get_extents_and_height (line=0x55555bf3bbc0, ink_rect=0x0, logical_rect=0x7fffffff2ae0, height=0x7fffffff2ad8) at ../pango/pango/pango-layout.c:5779
#15 get_line_extents_layout_coords (layout=layout@entry=0x55555bf347e0, line=0x55555bf3bbc0, layout_width=layout_width@entry=-1, y_offset=0, baseline=baseline@entry=0x7fffffff2b9c, line_ink_layout=line_ink_layout@entry=0x0, line_logical_layout=0x7fffffff2ba0)
    at ../pango/pango/pango-layout.c:2877
#16 0x00007ffff79def6c in pango_layout_get_extents_internal (layout=0x55555bf347e0, ink_rect=<optimized out>, ink_rect@entry=0x0, logical_rect=<optimized out>, logical_rect@entry=0x7fffffff2c10, line_extents=line_extents@entry=0x0) at ../pango/pango/pango-layout.c:3001
#17 0x00007ffff79df220 in pango_layout_get_extents (layout=<optimized out>, ink_rect=ink_rect@entry=0x0, logical_rect=logical_rect@entry=0x7fffffff2c10) at ../pango/pango/pango-layout.c:3118
#18 0x00007ffff79df26e in pango_layout_get_size (layout=<optimized out>, width=0x7fffffff2f00, height=0x7fffffff2e20) at ../pango/pango/pango-layout.c:3166
#19 0x00007ffff79ed827 in pango_layout_check_lines.part.0.lto_priv.0 (layout=<optimized out>) at ../pango/pango/pango-layout.c:4983
#20 0x00007ffff79df09f in pango_layout_check_lines (layout=<optimized out>) at ../pango/pango/pango-layout.c:4792
#21 pango_layout_get_extents_internal (layout=layout@entry=0x55555bf347e0, ink_rect=ink_rect@entry=0x0, logical_rect=logical_rect@entry=0x7fffffff3100, line_extents=line_extents@entry=0x0) at ../pango/pango/pango-layout.c:2925
#22 0x00007ffff79df220 in pango_layout_get_extents (layout=layout@entry=0x55555bf347e0, ink_rect=ink_rect@entry=0x0, logical_rect=logical_rect@entry=0x7fffffff3100) at ../pango/pango/pango-layout.c:3118
#23 0x00007ffff7e9648f in _pango_cairo_font_get_metrics (font=<optimized out>, language=<optimized out>) at ../pango/pango/pangocairo-font.c:293
#24 0x00007ffff79e9928 in pango_layout_get_empty_extents_and_height_at_index (layout=0x55555bfa6a30, index=0, logical_rect=0x7fffffff32e0, apply_line_height=apply_line_height@entry=1, height=height@entry=0x7fffffff32d8) at ../pango/pango/pango-layout.c:5557
#25 0x00007ffff79ea2db in pango_layout_get_empty_extents_and_height_at_index (layout=<optimized out>, index=<optimized out>, logical_rect=<optimized out>, apply_line_height=1, height=0x7fffffff32d8) at ../pango/pango/pango-layout.c:5493 [... cut ...]
#45 0x0000555556ba1e62 in Slic3r::GUI::GUI_App::on_init_inner() ()
#46 0x0000555556ba23fa in Slic3r::GUI::GUI_App::OnInit() ()
#47 0x0000555559a11627 in wxEntry(int&, wchar_t**) ()
#48 0x0000555556b622cb in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#49 0x0000555555fcafab in Slic3r::CLI::run(int, char**) ()
#50 0x0000555555ec8ba9 in main ()
```

To prevent this, it might be worth considering static-linking pango and cairo. The libraries shipped by Arch Linux seem to have ABI incompatibilities.

FFY00, 2024-09-20:
Also, I'd suggest avoiding recommending yay to folks, as it is a bit problematic. A much better/safer alternative would be [paru](https://github.com/Morganamilo/paru).

CitrusIntellect, 2024-09-25:
What is the problem with yay?

CitrusIntellect, 2024-09-25:
Just tried paru, much more helpful

closed 2025-01-01 by github-actions[bot]