OrcaSlicer/OrcaSlicer #6694: Skirt - draft shield loop problem

GlauTechCo, 2024-09-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

macOS

### OS Version

14.2.1 Sonoma

### Additional system information

Macbook Air M1

### Printer

FlashForge M5p

### How to reproduce

- Type the number of skirts other than one, for example 5
- activate draft shield

### Actual results

V 2.1.1
[image]

### Expected results

v 2.2.0 beta
[image]

### Project file & Debug log uploads

Ok

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kybarg, 2024-09-09:
not clear from the issue? is it expected to have 5 loops all the way or just a single loop?

GlauTechCo, 2024-09-09:
> not clear from the issue, is it expected to have 5 loops all the way or just a single loop?
> cause if you expect 1 loop on layers >1 then seems like it is fixed in 2.2.0 beta

[image]

As expected, there should be as many draft shield loops as skirt loops, right?

thunderkeys, 2024-11-04:
I would expect the prior behavior, or at least permit an option to have skirt loops the same on all layers.

thunderkeys, 2025-02-03:
is there any intent to change this behavior in a future release?

mobilinkd, 2025-02-24:
Please do not do this. The draft shield should have one loop.

thunderkeys, 2025-02-24:
I am not talking about the draft shield.

I am talking about if NOT using a draft shield, if you have skirt loops > 1 and z > 1, those should be respected across all layers.  At least as an option.

This allows for skirts to be easily picked off the bed.  This is how it works in every other slicer, until it was broken by supporting this draft shield feature request.

closed 2025-06-02 by github-actions[bot]