OrcaSlicer/OrcaSlicer #6699: Flushing volumes still influences multi-toolhead printers

Ataraxia-Mechanica, 2024-09-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0dev

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Voron 0 WIP tool changer

### How to reproduce

1. Enable "single extruder multi material"
2. Press the now appeared "flushing volume" button and set the values to something big
[image]
3. Disable "single extruder multi material"

### Actual results

[image]

### Expected results

flushing volume doesn't affect tool changer printers

### Project file & Debug log uploads

[Uploading OrcaCube.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

Ataraxia-Mechanica, 2024-09-09:
Related issue: https://github.com/SoftFever/OrcaSlicer/issues/6647

traxman25, 2024-09-09:
#6647 does not involved flushing volume. I have already set all flushing volumes to 0.

Ataraxia-Mechanica, 2024-09-09:
For me, this is 500 flushing:
[image]
This is 0 flushing:
[image]
So there are multiple things causing #6647 ... That's just great

traxman25, 2024-09-09:
And it just went away.... Changed to single, changed flushing multiplier to 0.1, changed back to multi tool and got the small prime line. Changed back to single, changed flushing multiplier to 0, changed back to multi tool. Slice and it had a large prime line again. When going back to multi tool each tool had changed the 'Extra length on start' to 0.2 so I changed that back to 0.1 like I had it before, sliced and the extra prime is gone. Saved the profile and the extra prime is gone.

I'm guessing there's too many "extra prime/purge/flush volume" settings in too many different places and in too many places that can only be adjusted in certain configurations which is causing the program to get mixed up.

staal54a, 2024-10-05:
I am also seeing this issue (2.2.0-beta2)

Igorshp, 2025-01-04:
can confirm this issue exists. The fix described (changing to "single extruder multimaterial" and editing flushing volumes) works.