OrcaSlicer/OrcaSlicer #6701: Positive part feature to exclude everything else

Rickthebig, 2024-09-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'am always frustrated when...
I have to put and align several negative parts in order to isolate the portion of the model that I wish to print.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Allow to add a *positive* part. This means that only the intersection of this part with the current model will be considered for printing. It would be a huge simplification.

kylek29, 2024-09-09:
I can see where that'd be very useful. As a workaround; if this is a model you made; you can create the negatives in the same CAD software and export them separately with the same origin point. That way you can do a simple import and set it to be negative.

Ataraxia-Mechanica, 2024-09-09:
In Blender this is called Boolean Intersect if you want to search this technique up.

Rickthebig, 2024-09-10:
> In Blender this is called Boolean Intersect if you want to search this technique up.

Thus proving that my idea is not so awkward.

I am even surprised that this has not been implemented yet, since it would save a lot of work creating a number of negative parts around, which would be shifted if reloaded (bug?) and would need to be re-aligned...

It is so obvious, and should not overburden the developers.

closed 2024-12-18 by github-actions[bot]