OrcaSlicer/OrcaSlicer #6704: Prime the extruder even if only a single extruder is used

Ataraxia-Mechanica, 2024-09-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

[image]
When "prime all printing extruders" is on and only using a single extruder, there aren't any priming lines. When this option is enabled, priming is usually disabled in the start g-code too, making this feature less useful.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Move "prime all printing extruders" feature out of the "prime tower" section, and make it work even when only a single extruder is used.

staal54a, 2024-09-13:
I think this would also be useful for toolchanger/IDEX printers. I have an IDEX printer myself and I don't use priming towers since I have purge buckets (or rather a place to park my nozzle over silicone so it doesn't ooze). So I don't want to waste filament on a priming tower but I would love to have an integrated priming option at the start of a print.

markdude701, 2024-10-26:
+1

The lack of this feature also appears to break P2PP support for the Palette multi color system for Orca Slicer as well. 
[image]

I think this would also help optimize some MMU functionality in my opinion, since that uses a single extruder multi color too.

closed 2025-02-01 by github-actions[bot]