OrcaSlicer/OrcaSlicer #6708: Max Bridge Length not visible in settings?

robert-stevenson-1, 2024-09-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Sovol SV06

### How to reproduce

1. In printer profile setting (Process panel), click the search icon
2. Search for "Max Bridge Length"
3. Click "Advance: Max Bridge Length"
4. Process panel move to "Support" tab, under "Advanced"
5. "Max Bridge Length" is not visible on the panel

### Actual results

Process panel move to "Support" tab, under "Advanced"
However, "Max Bridge Length" is not visible on the panel

### Expected results

"Max Bridge Length" is visible on the panel 
OR (If this setting is for internal use only/not intended for user edit) hide it in search

### Project file & Debug log uploads

Not relevant for this issue?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

[image]

Ergonomicmike, 2024-09-09:
I have noticed in Orca that sometimes a Search Term cannot be found because some other setting disallows that function.

But if you can't find the Search Term, you can't find the "help" that tells you that certain other settings disabled it.

I think I have asked, as an Enhancement, that ALL searchable items show, but are grayed out if disallowed at the time.

dingausmwald, 2024-09-18:
So how to make it visible? Searched a damn hour but couldn't find the setting

robert-stevenson-1, 2024-09-23:
> I have noticed in Orca that sometimes a Search Term cannot be found because some other setting disallows that function.
>
> But if you can't find the Search Term, you can't find the "help" that tells [...]

This would actually be a nice QoL behaviour change.

Also feel like I remember reading somewhere that the search feature also shows options that are purely accessible from the back end, but I'm not sure how true that statement is.

closed 2024-12-30 by github-actions[bot]