OrcaSlicer/OrcaSlicer #6719: Cannot create new filament: You have not selected a printer or preset yet. Please select at least one.

ThepHuck, 2024-09-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro Version	22H2 build 19045.4842

### Additional system information

Intel Core i7-4790k
16GB of RAM
GeForce GTX 1060 6GB
SSD hard drive

### Printer

Elegoo Neptune 4 Max running OpenNept4une

### How to reproduce

Click Filament drop down
Click Add/Remove filament
Click Custom Filaments tab
Click Create New
Provide Basic Information
Selected Create based on current filament (Same behavior if selecting "Copy Current Filament Preset")
Select existing filament
click create
[image]
[image]

### Actual results

error "You have not selected a printer or preset yet. Please select at least one." with only a "yes" button.

### Expected results

Create new filament preset.

### Project file & Debug log uploads

[debug_Mon_Sep_09_14_33_46_1512.log.zip](https://github.com/user-attachments/files/16948803/debug_Mon_Sep_09_14_33
[Nozzle temperature test.zip](https://github.com/user-attachments/files/16948825/Nozzle.temperature.test.zip)
_46_1512.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-09-11:
Duplicate.  There's half a dozen tickets for "can't create new filament"
I opened this one back in 2023
https://github.com/SoftFever/OrcaSlicer/issues/3242

markaudacity, 2024-10-04:
Still having this problem in October 2024.

KwisScram, 2024-11-09:
Me also in november 2024!!!

mdinnocenzo, 2024-11-12:
Currently on release 2.2.0 and I believe that it wants a printer to be selected.  I have noticed that when you try custom filaments it does not appear that the window is resizing to allow for complete selection of the options that pertain to this operation.

maindrive, 2025-02-15:
I'm having the same issue, I cant believe this has not been resolved yet, this is a big issue for me.

blockx, 2025-02-18:
Um, is it really still impossible to create a new filament in Orca 2.2.0 in 2025 ?  Is this receiving *any* attention?

WeevilKnight, 2025-02-19:
Same issue here. I'm on the latest, 2.2.0. Is anyone at support looking at this, or, are they stumped?

IsNotMyUserName, 2025-03-14:
It seems you are missing something here...
[image]

I get the same result "you have not selected a printer or preset yet..." When I try to export printer settings or configs from the "Preset" list

When I made a copy it ends up in "User presets" then there was no issue export stuff...

[image]

So look out for that hidden feature that Defaults are no good (branded shits) but your own profile is better.

Same goes for filaments, your own copies ends up in "user presets" vs "system presets"

Not very clear, but we are learning...

blockx, 2025-03-14:
> It seems you are missing something here... [image]
>
> I get the same result "you have not selected a printer or preset yet..." When I try to export printer settings or configs from the "Preset" list [...]

That response seems wholly impertinent ?

mikebryl, 2025-04-25:
You have to select the filament in "System Filaments" first. Now it is possible to create a custom filament profile.

[image]

steadmanc, 2025-05-16:
I recreated the above post.  I added the Generic Klipper Printer as a Custom Printer.  Then when I add a custom filament, it shows that generic printer as the only selectable option.  It does not show the default printers in that list.

gelse, 2025-06-02:
I have the same problem. Using KDE/wayland on a monitor with scaling. Rest of the info is in the screenshot.

[image]

Karl-Barx, 2025-06-22:
Dude, how old is this issue now? Crazy, maybe its time to get back on to prusa slicer idk

blockx, 2025-06-22:
> Dude, how old is this issue now? Crazy, maybe its time to get back on to prusa slicer idk

Yes, the responses here denying the existence of the problem really don't lend confidence in Orca development.  Before there is any chance an issue can be resolved it is necessary to acknowledge the issue exists...

cmyk, 2025-06-26:
What the heck? 2.3.1 on MacOS and this feature blocks me from adding custom filaments.
Seems the dialog box is cut off and has no scrolling.

[image]

blockx, 2025-06-26:
> What the heck? 2.3.1 on MacOS and this feature blocks me from adding custom filaments. Seems the dialog box is cut off and has no scrolling.
> 
> [image]

Yep.  It's f'ed, right enough...

cmyk, 2025-06-26:
> > What the heck? 2.3.1 on MacOS and this feature blocks me from adding custom filaments. Seems the dialog box is cut off and has no scrolling.
> > [image]
> 
> Yep. It's f'ed, right enough...

Most likely an easy fix. Lemme see....

ArrowThief, 2025-07-06:
I found a workaround for anyone still having this issue.
 
 1. Add a Generic filament.

[image]

2. Select the filament and then edit it. 

[image]

3. change settings as needed and save it. 

[image]

You should now have a custom filament you can change as needed. Don't forget to set compatible machines under the "Dependencies" tab. Otherwise the filament will only be available under the printer you had selected when creating it. 
Hope this helps someone.

armaguedes, 2025-07-14:
I've only ever used v2.3.0 on Windows 11, and I've always had this problem; the bottom of the "Create Filament" is never visible, and I have to resort to adding a generic filament, and then editing its parameters.

blockx, 2025-07-14:
> I've only ever used v2.3.0 on Windows 11, and I've always had this problem; the bottom of the "Create Filament" is never visible, and I have to resort to adding a generic filament, and then editing its parameters.

Yes, thanks Armaguedes, the problem does definitely exist.  I'm sure I speak for all users when I say that we are grateful for the work the developers and maintainers do in creating this great software for our hobby.   I understand that there is a need to prioritise and hope this will soon make it to the top of the tree in terms of getting fixed, it is very visible and does come up quite a bit.

Thanks again,

jbalius, 2025-07-26:
IMHO workaround solution, as this bug is unlikely to be fixed (bug is almost 2 years old already with no progress):

* In filaments select generic filament of same type
* Click Edit button on the right of selection
* In popup top right next to the same generic filament dropdown click save
* Popup will ask for a new name
* Type in your new name and save

The custom type filament will appear in you filament dropdown box. You can further edit it from there and set custom properties to it. I think this should reproduce same outcome as that buggy filament profile creation.

closed 2025-08-11 by SoftFever

MilitantTurtle, 2026-04-05:
Just came across this issue on latest release 2.3.2 stable, the filament manager in general seems to be broken, can't add a system preset or a custom preset.

commit e5d6de90 (2025-08-11): Fix inability to create custom filament due to hidden printer selection checkbox (#10362)

Fix clipped printer selection in Create Filament Preset dialog

Ensure the scrolled panel in the Create Filament Preset dialog expands vertically by changing `m_main_sizer->Add(..., proportion=0)` to `proportion=1`. This exposes the printer selection checkbox, which is required to create a custom filament.

Fixes #6719