OrcaSlicer/OrcaSlicer #6723: Virtual Camera install fails on fresh install of nightly build 2.2.0-beta

msteele999, 2024-09-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu Lab X1

### How to reproduce

- uninstall existing version of OrcaSlicer and Bambu Studio
- delete c:\users\myuser\appdata\roaming\orcaslicer directory
- install nightly OrcaSlicer build
- complete wizard setup including Bambu network setup
- authenticate with google and sync preset, filaments, etc.
- go to Device
- go to Camera Settings
- select Go Live
- asked to install the virtual camera tools
- error: "Failed to install the plug-in. Please check whether it is blocked or has been deleted by anti-virus software."
- I've attempted to install with Windows Defender disabled with not change.

### Actual results

Fails to install the virtual camera tools

### Expected results

Should install the virtual camera tools

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16951200/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

msteele999, 2024-09-10:
Update: I am not able to start the camera either - "Player is malfunctioning. Please reinstall the system player [2:2]"

msteele999, 2024-09-10:
Update: If I install the latest Bambu Studio, the watch camera in OrcaSlicer begins working and warns that it needs to register a DLL for Orcaslicer, which it does. Additionally Go Live does not prompt for the virtual camera tools - however there is still no cameratools directory created in the OrcaSlicer directory.

There seems to be some cross pollination happening between OrcaSlicer and Bambu Studio 

Here is another log summary:
[log (2).zip](https://github.com/user-attachments/files/16954173/log.2.zip)

SturmB, 2024-09-27:
I have this same issue on a fresh install of the stable 2.1.1 version.

It seems to be complaining about networking tools, even though those are already installed just fine; it's the _camera_ tools that need to be installed.

Thankfully, I still have Bambu Studio installed with the camera tools, so I was able to perform the following steps and get it working in OrcaSlicer:

1. Shut down OrcaSlicer
2. Copy over the `cameratools` directory and its contents from `%APPDATA%\BambuStudio` to `%APPDATA%\OrcaSlicer`
3. Start up OrcaSlicer and try enabling "Go Live" again

johann-miller, 2024-10-27:
> Thankfully, I still have Bambu Studio installed with the camera tools, so I was able to perform the following steps and get it working in OrcaSlicer:
>
>     1. Shut down OrcaSlicer
>
>     2. Copy over [...]

Thank you! I was having this same issue with OS 2.1.1 and this fixed it.

closed 2025-02-02 by github-actions[bot]