OrcaSlicer/OrcaSlicer #6729: Temp Tower calibration does not change temp.

bibidentuhanoi, 2024-09-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1, 2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

AnyCubic Kobra 2 Neo running Klipper

### How to reproduce

1.  Go to Calibration opetion
2. Choose Temparature
3. Click Ok 
4. Get a Gcode file
5. Print

### Actual results

The temparature stays at 230c the whole time 
all the M104 gcode commands  are M104 S220 ; set nozzle temperature

### Expected results

The Temp need to change. 

### Project file & Debug log uploads

[error]	2024-09-11 12:46:52.089950[Thread 0x00002a54]:can not find parent for config C:\Users\BarryAllen\AppData\Roaming\OrcaSlicer\user\default\process\0.20mm Standard @MyKlipper zamonary1.json!
[error]	2024-09-11 12:46:52.090948[Thread 0x00002a54]:can not find parent for config C:\Users\BarryAllen\AppData\Roaming\OrcaSlicer\user\default\process\Anycubic kobra 2 neo klipper 0.2mm.json!
[error]	2024-09-11 12:46:52.093950[Thread 0x00002a54]:can not find parent for config C:\Users\BarryAllen\AppData\Roaming\OrcaSlicer\user\default\machine\Anycubic kobra 2 neo klipper 0.2mm.json!
[error]	2024-09-11 12:46:52.094949[Thread 0x00002a54]:can not find parent for config C:\Users\BarryAllen\AppData\Roaming\OrcaSlicer\user\default\machine\MyKlipper 0.4 nozzle - Copy.json!
[error]	2024-09-11 12:46:55.238582[Thread 0x00002a54]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-09-11 12:46:58.463450[Thread 0x00002a54]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[temperature_tower_PLA_0.15_2h29m.txt](https://github.com/user-attachments/files/16958520/temperature_tower_PLA_0.15_2h29m.txt)

tlhintoq, 2024-09-12:
I will add....

It shouldn't use `M104 S` as that has no wait to get to the new target temperature.  So your town tiers are gradients and hard to read.

It should use `M104 R` directing it to wait for the new target temperature even when decreasing.

Most temp tower generators suffer this same problem and its annoying that that all just copy each other.

https://marlinfw.org/docs/gcode/M109.html

shirgall, 2024-09-12:
> I will add....
> 
> It shouldn't use `M104 S` as that has no wait to get to the new target temperature. So your tower tiers are gradients and hard to read instead of being strict tiers with obvious changes at the start of each one.
> 
> It should use `M104 R` directing it to wait for the new target temperature even when decreasing.
> 
> Most temp tower generators suffer this same problem and its annoying that that all just copy each other.
> 
> https://marlinfw.org/docs/gcode/M109.html

Can confirm that there's M104 S195 for this layer change here. Perhaps open a separate issue?
[image]

closed 2024-12-20 by github-actions[bot]