OrcaSlicer/OrcaSlicer #6733: Wrong line type when printing supports in 2 different materials

Ataraxia-Mechanica, 2024-09-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0dev

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Voron 0 WIP tool changer

### How to reproduce

Use 2 different materials for support, then a layer of support interface is marked as support, causing it to be printed vary fast and fail. 
[image]
[image]
[image]

### Actual results

Print fails as support sticks onto the nozzle.

### Expected results

This layer is marked as support interface instead, slowing it down and allowing it to be printed properly.

### Project file & Debug log uploads

[Hilbert_2it_Square.zip](https://github.com/user-attachments/files/16963925/Hilbert_2it_Square.zip)
The first plate shows this clearly

### Checklist of files to include

- [ ] Log file
- [X] Project file

Theaninova, 2024-11-01:
This occurs when the support structure has to switch pattern, aka has to support itself. It will correctly lay out one layer of support base material, but then regardless of which parts touch the actual print also follow it up with two layers of interface material.

This is definitely a big issue for using support materials combinations like PETG/PLA because they don't stick to each other, and very small supports on top of these interfaces will very quickly peel off...

(dark green is printed in the interface material, light green in the base material)

[image]
[image]
[image]
[image]

Ataraxia-Mechanica, 2025-02-03:
pls fix

Ataraxia-Mechanica, 2025-05-05:
no

Ataraxia-Mechanica, 2026-04-01:
> This is a structure from Bambu Studio called `support transition layer`. However it doesn't have proper support in OrcaSlicer, just a ported feature. I personally would like to disable it entirely.

I want to see it get printed at a configurable speed or support interface speed, as it should be useful for preventing supports separating from support interface when they are printed slowly, instead of the same speed as supports