OrcaSlicer/OrcaSlicer #6735: Crash on starting up, no config created

ilyan1n, 2024-09-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Additional system information

CPU: AMD Ryzen 5 3600
RAM: 32 GB
Dispay Card: NVIDIA GeForce GTX 1660

### Printer

Creality Ender 3

### How to reproduce

1. Start up program

### Actual results

Trying to load configuration, then crash without any error windows

### Expected results

Start up program

### Project file & Debug log uploads

Crash log:

Exceptions:

Exception Code :c0000005 ACCESS_VIOLATION

Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x0 
Context :000000ED2F5291C0 
ContextFlag : 0x10005f, EFlags: 0x10206 
Fault address:  0x72CF4164 0x1:0x2113164 C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:
7ffc72cf4164:hid_write [Unknown file][-1]
7ffc72cf231d:hid_write [Unknown file][-1]
7ffc72cf1fac:hid_write [Unknown file][-1]
7ffc7213647f:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffc7209cf95:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffc720883e1:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffc72088aa2:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffc71f4b589:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffc71f48c7c:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffc71afb32b:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc71ae132e:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc718d4325:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc718ae359:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc72bec2e9:hid_write [Unknown file][-1]
7ffc72c49a58:hid_write [Unknown file][-1]
7ffc71886cb8:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffc70f38b42:Unknown [Unknown file][-1]
7ffc70fa65e2:orcaslicer_main [Unknown file][-1]
7ff67f5159d5:Unknown [Unknown file][-1]
7ff67f515e96:Unknown [Unknown file][-1]
7ffd0be4257d:BaseThreadInitThunk [Unknown file][-1]
7ffd0d9caa48:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x72CF4164 0x1:0x2113164 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x72CF231D 0x1:0x211131D C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x72CF1FAC 0x1:0x2110FAC C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x7213647F 0x1:0x155547F C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x7209CF95 0x1:0x14BBF95 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x720883E1 0x1:0x14A73E1 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x72088AA2 0x1:0x14A7AA2 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x71F4B589 0x1:0x136A589 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x71F48C7C 0x1:0x1367C7C C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x71AFB32B 0x1:0xF1A32B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x71AE132E 0x1:0xF0032E C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x718D4325 0x1:0xCF3325 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x718AE359 0x1:0xCCD359 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x72BEC2E9 0x1:0x200B2E9 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x72C49A58 0x1:0x2068A58 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x71886CB8 0x1:0xCA5CB8 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x70F38B42 0x1:0x357B42 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x70FA65E2 0x1:0x3C55E2 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x7F5159D5 0x1:0x49D5 C:\Program Files\OrcaSlicer\orca-slicer.exe
0x7F515E96 0x1:0x4E96 C:\Program Files\OrcaSlicer\orca-slicer.exe
0xBE4257D 0x1:0x1157D C:\Windows\System32\KERNEL32.DLL
0xD9CAA48 0x1:0x59A48 C:\Windows\SYSTEM32\ntdll.dll

Debug log: Empty

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Tried anything what I could imagine. I have no idea what to do

ilyan1n, 2024-09-11:
Tried different version. Downgrade, beta, portable - nothing works

FlailAway, 2024-09-11:
Me too, but with Linux Mint-22+Cinnamon desktop. Mint-22 uses Ubuntu-24.04. reboots instantly.

closed 2024-12-19 by github-actions[bot]