OrcaSlicer/OrcaSlicer #6736: Precise Z-height on all top most surfaces

kybarg, 2024-09-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently setting Precise Z-height only adjusts the last layer of the print (green line) while other top most surfaces (red lines) layers are not adjusted .
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I think it sohuld take all the **top most** surfaces into account or we shoudl have additional checkbox to count for that.

### Additional context

I'm working on technical parts where precision matters and there are cases wehre it is impossible to model part with layer height in mind.

Sidiusz, 2024-09-12:
That's a good suggestion

vgdh, 2024-09-12:
Great idea!

vgdh, 2024-12-12:
up