OrcaSlicer/OrcaSlicer #6739: Crash on launch (Locale issue?)

goktug7913, 2024-09-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1-1

### Operating System (OS)

Linux

### OS Version

Linux archlinux 6.10.9-arch1-2 #1 SMP PREEMPT_DYNAMIC Tue, 10 Sep 2024 14:37:32 +0000 x86_64 GNU/Linux

### Additional system information

CPU: Intel i7-6700
GPU: ASUS TUF RTX 3070 (nvidia-dkms)
RAM: Corsair Vengeance 32GB DDR4 3600Mhz

### Printer

None but normally Ender 3

### How to reproduce

yay orca-slicer-bin
orca-slicer

### Actual results

```
bishop@archlinux ~> orca-slicer
[2024-09-12 14:54:21.215627] [0x000079f787f25c00] [trace]   Initializing StaticPrintConfigs
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
fish: Job 1, 'orca-slicer' terminated by signal SIGABRT (Abort)
```

### Expected results

Orca slicer starting

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2024-09-12:
Thanks for the update.
I've seen quite a lot application crashes if there is only LANG=C available.

It's likely an OS setup problem to me.  As application with localization support need these.
But I must admit I don't bother to digger into this issue.

goktug7913, 2024-09-12:
I got it running, however the 3S viewport is missing from the UI now.

I also got a segfault on launch after fixing the locale problem which is caused by having noto sans installed (it is a dependency of orca-slicer) force removing the noto package fixes that issue.

namtabmai, 2024-09-13:
You seem to be on arch, try downgrading the ttf-harmonyos-sans. The newer version of the font appears to be causing a crash.

Appears that going from: https://developer.huawei.com/images/download/general/HarmonyOS-Sans.zip
to
https://developer.huawei.com/images/download/general/HarmonyOS-Sans.zip

is an issue. Not sure how loading a font can cause a segfault.

Stacktrace

`
#0  0x000055555913de74 in tt_face_get_paint_layers ()
#1  0x00007ffff716c4de in _cairo_ft_scaled_glyph_is_colr_v1 (scaled_font=<optimized out>, scaled_glyph=0x55555c0bf9a0, face=0x55555c0bea10) at ../cairo/src/cairo-ft-font.c:3194
#2  _cairo_ft_scaled_glyph_init_metrics (scaled_font=<optimized out>, scaled_glyph=0x55555c0bf9a0, face=0x55555c0bea10, vertical_layout=0, load_flags=66048, foreground_color=0x7ffff71d9c40 <cairo_color_black.lto_priv>) at ../cairo/src/cairo-ft-font.c:3268
#3  _cairo_ft_scaled_glyph_init (abstract_font=<optimized out>, scaled_glyph=0x55555c0bf9a0, info=CAIRO_SCALED_GLYPH_INFO_METRICS, foreground_color=0x7ffff71d9c40 <cairo_color_black.lto_priv>) at ../cairo/src/cairo-ft-font.c:3381
#4  0x00007ffff711d9ad in _cairo_scaled_glyph_lookup (scaled_font=scaled_font@entry=0x55555c0bf460, index=<optimized out>, info=info@entry=CAIRO_SCALED_GLYPH_INFO_METRICS, foreground_color=0x7ffff71d9c40 <cairo_color_black.lto_priv>, 
    foreground_color@entry=0x0, scaled_glyph_ret=scaled_glyph_ret@entry=0x7fffffff51a0) at ../cairo/src/cairo-scaled-font.c:2913
#5  0x00007ffff711dc83 in cairo_scaled_font_glyph_extents (scaled_font=0x55555c0bf460, glyphs=glyphs@entry=0x7fffffff51f0, num_glyphs=num_glyphs@entry=1, extents=extents@entry=0x7fffffff5210) at ../cairo/src/cairo-scaled-font.c:1643
#6  0x00007ffff7ea5242 in compute_glyph_extents (cf_priv=0x55555c12e908, glyph=28635, entry=0x55555c1320b8) at ../pango/pango/pangocairo-font.c:807
#7  _pango_cairo_font_private_get_glyph_extents_cache_entry (cf_priv=0x55555c12e908, glyph=28635) at ../pango/pango/pangocairo-font.c:833
#8  _pango_cairo_font_private_get_glyph_extents (cf_priv=0x55555c12e908, glyph=28635, ink_rect=0x7fffffff52c0, logical_rect=0x7fffffff52d0) at ../pango/pango/pangocairo-font.c:870
#9  pango_cairo_fc_font_get_glyph_extents (font=0x55555c12e890, glyph=28635, ink_rect=0x7fffffff52c0, logical_rect=0x7fffffff52d0) at ../pango/pango/pangocairo-fcfont.c:130
#10 0x00007ffff79d0779 in pango_glyph_string_extents_range (glyphs=0x55555b3e30c0, start=0, end=5, font=0x55555c12e890, ink_rect=<optimized out>, logical_rect=0x7fffffff5380) at ../pango/pango/glyphstring.c:210
#11 pango_glyph_string_extents_range (glyphs=0x55555b3e30c0, start=0, end=5, font=0x55555c12e890, ink_rect=<optimized out>, logical_rect=0x7fffffff5380) at ../pango/pango/glyphstring.c:164
#12 0x00007ffff79e901a in pango_layout_run_get_extents_and_height (run=0x55555b3e2f50, run_ink=run_ink@entry=0x7fffffff5460, run_logical=0x7fffffff5380, run_logical@entry=0x0, line_logical=line_logical@entry=0x7fffffff5450, height=height@entry=0x7fffffff544c) at ../pango/pango/pango-layout.c:5640
#13 0x00007ffff79ea253 in pango_layout_line_get_extents_and_height.part.0.lto_priv.0 (line=0x55555c002930, ink_rect=<optimized out>, logical_rect=<optimized out>, height=<optimized out>) at ../pango/pango/pango-layout.c:5836
#14 0x00007ffff79deb46 in pango_layout_line_get_extents_and_height (line=0x55555c002930, ink_rect=0x0, logical_rect=0x7fffffff54e0, height=0x7fffffff54d8) at ../pango/pango/pango-layout.c:5779
#15 get_line_extents_layout_coords (layout=layout@entry=0x55555c08a9e0, line=0x55555c002930, layout_width=layout_width@entry=-1, y_offset=0, baseline=baseline@entry=0x7fffffff559c, line_ink_layout=line_ink_layout@entry=0x0, line_logical_layout=0x7fffffff55a0) at ../pango/pango/pango-layout.c:2877
#16 0x00007ffff79def6c in pango_layout_get_extents_internal (layout=0x55555c08a9e0, ink_rect=<optimized out>, ink_rect@entry=0x0, logical_rect=<optimized out>, logical_rect@entry=0x7fffffff5610, line_extents=line_extents@entry=0x0) at ../pango/pango/pango-layout.c:3001
#17 0x00007ffff79df220 in pango_layout_get_extents (layout=<optimized out>, ink_rect=ink_rect@entry=0x0, logical_rect=logical_rect@entry=0x7fffffff5610) at ../pango/pango/pango-layout.c:3118
#18 0x00007ffff79df26e in pango_layout_get_size (layout=<optimized out>, width=0x7fffffff5900, height=0x7fffffff5820) at ../pango/pango/pango-layout.c:3166
#19 0x00007ffff79ed827 in pango_layout_check_lines.part.0.lto_priv.0 (layout=<optimized out>) at ../pango/pango/pango-layout.c:4983
#20 0x00007ffff79df09f in pango_layout_check_lines (layout=<optimized out>) at ../pango/pango/pango-layout.c:4792
#21 pango_layout_get_extents_internal (layout=0x55555c08a9e0, ink_rect=ink_rect@entry=0x0, logical_rect=logical_rect@entry=0x7fffffff5ad0, line_extents=line_extents@entry=0x0) at ../pango/pango/pango-layout.c:2925
#22 0x00007ffff79e31b4 in pango_layout_get_pixel_size (layout=<optimized out>, width=0x7fffffff5b44, height=0x7fffffff5b48) at ../pango/pango/pango-layout.c:3194
#23 0x000055555991d888 in wxCairoContext::GetTextExtent(wxString const&, double*, double*, double*, double*) const ()
#24 0x00005555596bf9d7 in wxGCDCImpl::DoGetTextExtent(wxString const&, int*, int*, int*, int*, wxFont const*) const ()
#25 0x0000555556b7429a in Slic3r::GUI::SplashScreen::Decorate(wxBitmap&) ()
#26 0x0000555556b69dc5 in Slic3r::GUI::GUI_App::on_init_inner() ()
#27 0x0000555556b6b62a in Slic3r::GUI::GUI_App::OnInit() ()
#28 0x00005555599cd4d7 in wxEntry(int&, wchar_t**) ()
#29 0x0000555556b2a01b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#30 0x0000555555fc442b in Slic3r::CLI::run(int, char**) ()
#31 0x0000555555ec3209 in main ()
`

goktug7913, 2024-09-13:
@namtabmai i will do your suggestion, thanks.

Removing it also seems to fix it, however I cannot use orca slicer even if it starts. The 3D viewport UI element is simply not drawn. I will provide a screenshot when I get on my PC.

I also saw some GTK errors, not sure if it was related.

nikoraasu, 2024-09-14:
The UI not rendering is an Nvidia driver issue in my case, I had to downgrade from 560 to 555, though this is a temporary "band-aid" solution.

Zunon, 2024-09-19:
I've got the same issue on a japanese locale in endeavourOS with orca-slicer-git on the latest version
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
中止 (コアダンプ)
```

DerCakeman, 2024-11-20:
For me the issue was that my environmental variable was set to "LC_NUMERIC=en_DE.UTF-8", which doesn't really exist and is a bug within KDE Plasma i think. I was able to start it by just setting it to "LC_NUMERIC=en_US.UTF-8".

closed 2025-02-26 by github-actions[bot]