OrcaSlicer/OrcaSlicer #6740: Arrange All Objects not working on beta nightly build

patofoto, 2024-09-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 beta Nightly Build

### Operating System (OS)

macOS

### OS Version

macOS 14.6.1

### Printer

Voron 2.4 350

### How to reproduce

1. Place object on plate
2. clone several
3. click on arrange all objects button on top of workspace
4. type 10 in spacing box
5. click on arrange

### Actual results

Objects don't space at all. Only if 0 is used as a value in the spacing box.

### Expected results

Objects on plate should space out properly to distance determined on spacing box

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16980683/log.zip)
[Cube.3mf.zip](https://github.com/user-attachments/files/16980684/Cube.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

SoftFever, 2024-09-12:
Thank you!
It should have been fixed now.

closed 2024-09-12 by SoftFever

patofoto, 2024-09-12:
Thanks. Compiling to test now.

patofoto, 2024-09-12:
Confirmed. Working again. Thanks.