OrcaSlicer/OrcaSlicer #6743: Read a multi-body STEP as multiple bodies in their assembles orientation (for integrated light pipes)

W1SPS, 2024-09-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not so much a problem, just a useful feature I think doesn't exist. I would like a multi-body STEP file (saved from an assembly in CAD software) to be read in and acknowledged as such so that integrated light pipes, through text, or symbols could be made of a translucent material. This way light sources inside an enclosure can be shown through an enclosure cover for instance.  The benefit would be that the enclosure cover (and accompanying clear sections would be seamless and have the same surface texture throughout.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I think if a multi body part was read in with all parts in their assembled orientation and the slicer asked: "Is this a multi body part?" You answer yes and you don't get warnings about parts too close and such. This would allow you to not only make areas diffused light can shine through, but also allow you to make integrated light pipes to light sources far below the enclosure cover.

### Describe alternatives you've considered

No Idea I can think of addresses the way I wish this could be handled within the software.

W1SPS, 2024-09-12:
So... In this picture the translucent "Light Pipe" would be assembled in the CAD software (In my case Solidworks). Then Orcaslicer reads the STEP file keeping the Assemblies' parts in their assembled orientation. Then a translucent material is chosen for the "Light Pipe" and an opaque material is chosen for the main part (Such as transparent PETG and PETG-CF). Since I would print this face down the finished part should have an even, unbroken texture and would be essentially a single part after printing.

[image]

xokahag, 2024-10-19:
You can close your own issue report by typing a comment and then click "comment and close". Helps keeping the backlog clean :)

closed 2025-01-25 by github-actions[bot]