OrcaSlicer/OrcaSlicer #6755: 【BUG】Prime Tower dangling appears. a series of bugs.

funnygeeker, 2024-09-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-Beta

### Operating System (OS)

Windows

### OS Version

Windows11

### Additional system information

i5-12450 RAM 16GB

### Printer

Snapmaker J1s

### How to reproduce

Add model：3DBenChy
[image]
[image]

when upgrading from some old versions（v2.1.1） to new versions. The amount of material cleaning in the prime tower may not be correct.
prime tower will be higher prime volume.
[image]
[image]

（Please forgive me for using translation software to translate these contents. ThenI don't have enough time now, so I can only briefly report it, but I can't provide more detailed information.I'll add it in the follow-up. I'm very sorry.）

### Actual results

[image]

### Expected results

.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2024-09-15:
Thanks
Can you attach a sample project to reproduce the issue?

funnygeeker, 2024-09-16:
> 谢谢您能否附上示例项目来重现该问题？

Okay, I have time today. The file is as follows:
[BUG_3DBenchy.zip](https://github.com/user-attachments/files/17011772/BUG_3DBenchy.zip)

funnygeeker, 2024-09-30:
@SoftFever It seems that you forgot to reply. I tested it again in the latest Beta2, but the problem persists. 
[BUG_3DBenchy.zip](https://github.com/user-attachments/files/17011772/BUG_3DBenchy.zip)

commit 7736cd1b (2024-10-06): fixed #6755
PR #6980 (2024-10-07): Fix an issue that wrong z value was used if a new layer is started with wipe tower extrusions.
…

# Description

This update, along with #6934, resolves the remaining issues with the wipe tower.

The GCode class will no longer maintain its own Z height information (m_nominal_z). Instead, it will utilize the Z height from GCodeWriter's m_pos. This approach is less error-prone.

This PR also fixes #6755.
It also enforce "Enable Filament ramming" option

@igiannakas It should fix the wipe tower crashing issue
# Screenshots/Recordings/Graphs

## Tests
commit ab55d4ee (2024-10-07): Fix an issue that wrong z value was used if a new layer is started with wipe tower extrusions. (#6980)

…

# Description

This update, along with #6934, resolves the remaining issues with the
wipe tower.

The GCode class will no longer maintain its own Z height information
(m_nominal_z). Instead, it will utilize the Z height from GCodeWriter's
m_pos. This approach is less error-prone.

This PR also fixes #6755.
It also enforce "Enable Filament ramming" option

@igiannakas It should fix the wipe tower crashing issue
# Screenshots/Recordings/Graphs

## Tests
closed 2024-10-07 by SoftFever