OrcaSlicer/OrcaSlicer #6758: Extrusion rate smoothing does not work when moving to another island

DanDan2000-1, 2024-09-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Extrusion rate smoothing is ON:
max_volumetric_extrusion_rate_slope = 5mm^3/s^2
max_volumetric_extrusion_rate_slope_segment_length = 3
This works well as long as you don't go off-island.
In the case when we have several islands (several parts), when moving the extruder to another island, the speed іьщщерштп does not work at all. As a result, the new permeter starts printing at a different speed, which has changed dramatically. This leads to overextrusion at the beginning of the perimeter.
[image]

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

Before moving to another island, the extruder must gradually change its speed to match the speed of the extruder on the new island.

vgdh, 2024-09-14:
Looks more like a bug

DanDan2000-1, 2024-09-17:
The same problem exists when moving to a new layer. I think this feature is not fully done yet.

closed 2024-12-25 by github-actions[bot]

Rickthebig, 2025-11-23:
Please re-open this bug report.
I am currently struggling against sudden extrusion changes between loops (islands) in the same layer, when using TPU. When jumping from a low loop on a small island to a large loop of a large island (in the same layer), I get underextrusion. The other way around I get overextrusion. TPU is specially susceptible to this issue and demands inconvenient workarounds and reprints. I consider this a clear bug.