OrcaSlicer/OrcaSlicer #6760: Prime tower grew very large after slicing

goombah88, 2024-09-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Linux_Ubuntu2404_V2.2.0-beta

### Operating System (OS)

Linux

### OS Version

EndeavourOS (arch linux)

### Additional system information

CPU: AMD Ryzen 9 7950X 16-Core
MEM: 32 GB
Video Card: AMD Radeon RX 7900 XTX

### Printer

Bambu P1S

### How to reproduce

I added a cylinder to the build plate.
Add smaller cylinder centered.
Changed smaller cylinder to a negative part
Sliced

### Actual results

Prime tower grew in the Y direction to go off the top and bottom of the build plate, giving a gcode error as well.

### Expected results

I have my prime tower set to
Width: 35 mm
Prime Volume: 45 mm3
Brim Width: 3 mm

I never change those settings. So the prime tower should not have changed at all when slicing a model. I've never seen this issue before on any previous versions of Orcaslicer. I thought I fixed it by disable/enable the prime tower. That reset the size and put the tower in the lower right corner of the build plate (previously I had it in the upper right corner, if relevant).
But now each time I slice it grows all over again.

### Project file & Debug log uploads

[fan_blades.zip](https://github.com/user-attachments/files/17002556/fan_blades.zip)
[debug_.zip](https://github.com/user-attachments/files/17002569/debug_.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

goombah88, 2024-09-14:
Might be related to https://github.com/SoftFever/OrcaSlicer/issues/6755 but it's hard to be sure since that bug doesn't have much info in it to go on.

vgdh, 2024-09-14:
Preview size almost always incorrect.
You can make it smaller by decreasing the amount of **prime volume**

vgdh, 2024-09-14:
Also try to decrease multiplier in **Flushing volume** section

goombah88, 2024-09-14:
> Also try to decrease multiplier in **Flushing volume** section

I hear ya, but it still has to be a bug with the beta because I have never seen this issue before on any prior version of Orca (or any other slicer for that matter).
[image]
Again, unchanged from previous versions. Untouched for quite a while actually.

SoftFever, 2024-09-15:
Thanks.
I can repro it.

Will look into it

SoftFever, 2024-09-15:
Confirmed it's bug caused by the "Precise Z height" option.
It's not a regression bug in 2.2.0-beta.

For now, please disable it in this option as a workaround.

goombah88, 2024-09-15:
Confirmed. The workaround removes the issues. Thanks for the quick turnaround. I'll do some more beta testing today.

KelsamGames, 2024-09-16:
This bug still exists for me whether the Precise Z height option is enabled or not. Prime tower looks right in Prepare, but then generates a huge one when sliced.

vanowm, 2024-12-16:
Can someone please remove this useless bot, just because there is no activity on an issue, it doesn't make it fixed...(referring to closed by the bot issue)

jctune, 2024-12-16:
Just noting that I encountered this bug only recently after updating to the 2.3.0-dev released approx. 12/14/25.  Had not encountered this bug previously.

The bug is occurring even though I do not have "Precise Z height" enabled.

vgdh, 2024-12-16:
> Can someone please remove this useless bot, just because there is no activity on an issue, it doesn't make it fixed...(referring to closed by the bot issues)

Be patient. The purpose of the bot automatically close all not relevant issues. If issue is relevant like this one, people will continue to bump it up.

SoftFever, 2024-12-22:
Hey guys,

Can you share the sample project if you experienced this issue even though you have the precise Z is not disabled?

Apotrox, 2025-01-19:
Hey @SoftFever, got one for ya.

[image]

[multimat_issue.zip](https://github.com/user-attachments/files/18469334/multimat_issue.zip)

Also does funny passes at the front?

SoftFever, 2025-02-23:
This should be fixed in 2.3.0-beta.

closed 2025-02-23 by SoftFever

githomps, 2025-02-26:
> This should be fixed in 2.3.0-beta.

Hi @SoftFever, still encountering the same problem. I didn't have Precise Z height enabled when the issue initially presented itself, and enabling it didn't have any impact. 

The issue occurs when using variable layer height after slicing the model. The length of the tower changes depending on the value chosen for the Quality/Speed setting. Applying a 0.5 caused the tower to massively extend in length, easily over a foot. Reducing the value to 0.2 and reslicing caused the tower to shrink to ~6" long, still massive but within the bounds of the build plate. Increasing the value to 0.75 and reslicing returned the tower to its default size. 

I initially encountered the issue in version 2.2.0 attempting to slice a file that I had previously printed with prime tower and variable layer height enabled with no problems. I hadn't saved the file before printing the first time, and Orca crashed, losing my updates. I attempted to re add the models and update my support settings, when the bug presented itself. I updated to the beta version 2.3.0, and tried making a new project using the same model, but the problem persisted. 

Screenshots below after slicing at Variable Layer Height Quality/Speed setting 0.5, reslicing at 0.2, again at 0.75, and then after resetting Variable Layer Height. I also attached the project file.

0.5
[image]
0.2
[image]
0.75
[image]
Reset
[image]

[Prime Tower Issue File.zip](https://github.com/user-attachments/files/18980553/Prime.Tower.Issue.File.zip)

Aggrajag68, 2025-03-20:
I still see this problem with 2.3.0 release build 70931e5. It's easy to reproduce:
1) Start a new project.
2) Add any model (I used Benchy).
3) Enable Prime Tower.
4) Click Slice Plate and all is well and good. No tower required.
5) Move the layer slicer down to e.g. slice 174.
6) Add Filament change and re-slice.
7) Looks good but there's a pretty huge tower, no need to waste so much.
8) Change tower Width to 10 (which seems reasonable) and re-slice.
9) Tower extends way off the table.
10) There is no volume option visible (although it appears in the search) so this is not changeable.

jamesche75, 2025-06-10:
I can confirm this issue still happens in 2.3.0, Build 70931e5. If I disable Precise Z Height, the issue goes away. However, I do see that some people report this even when that option is disabled, so there may be multiple issues here. 

By the way, I have been using Precise Z on this project since installing 2.3.0 without issue. The problem suddenly popped up today with no changes I'm aware of.

zoran-grujic, 2025-06-14:
In my 2.3.0 Build 70931e5  the bug persists regardless on Precise Z Height or Precise wall. For the first few slice occasions it was OK, and now every new slicing the prime tower is huge in size. I manege to make it reasonable by setting Flushing volumes as @goombah88 suggested. 

[image]