OrcaSlicer/OrcaSlicer #6761: Configure Leyer Height by Heigth

feliperiffel, 2024-09-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I would like a way to configure the quality of different `Leyer Heigth` by height in mm.

This would be useful to improve printing time without necessarily affecting quality, for example, I have some gears to print I don't care how high it prints in the bottom or middle of the gear but for the teeth, I would like to decrease the layer height to better precision/strength.

Take this example:
In this gear, if I use 0.2 layer height the teeth of the gear are in the last 3 layers of the print
[image]

if I change it to 0.08 it has 8 layers, 
[image]

I know I could print all in 0.08, but if I could say that from the base to 6mm I would like a layer height of 0.2, and just the last 0,06 mm use a 0.08 layer hight I would be able to have a faster printer without necessarily mess with the quality.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

In the left panel, it would be nice to have the ability to break the layer height by global height, giving the possibility to in the same print I have different qualities at the same time.

[image]
to

henrivdr, 2024-09-15:
Have you tried the Variable Layer Height option?

SoftFever, 2024-09-15:
You can use Height Range Modifier  
[image]

 or Variable Layer height tool to do this. 
[image]

closed 2024-09-15 by SoftFever

feliperiffel, 2024-09-15:
Thank you! I searched on the web before sending the request but really did not find it.