OrcaSlicer/OrcaSlicer #6763: Unable to add PAUSE in Preview screen slider. Only Jump To is available...

rainmakerz17, 2024-09-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Creality K1

### How to reproduce

Slice a model.  Got into Preview screen.  Slide the layer selector down and right click.  The Add PAUSE function will not work now.  I used this function yesterday just fine on the 2.1.x version of Orca...

### Actual results

Unable to Add PAUSE to the gcode

### Expected results

I expect to be able to Add PAUSE functionality to my model as I could yesterday using version 2.1.x.

### Project file & Debug log uploads

[thug life glasses - 6545959.zip](https://github.com/user-attachments/files/17002970/thug.life.glasses.-.6545959.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Nothing else to add

rainmakerz17, 2024-09-14:
UPDATE: Issue resolved.  I reset my Process profile to the standard and this resolved my issue.  I am not sure what setting I had changed in my custom Process profile that caused the PAUSE to be voided, but just glad that I am still able to use this function.  
Pleases remove this bug ticket.  TY

vgdh, 2024-09-14:
you can close the issue by comment it and choose **comment and close**

rainmakerz17, 2024-09-14:
RESOLVED.  I deleted my Process profile and used the standard process profile and this reset my configurations to defaults.  I must have had some combination of settings made that did not allow the Pause to function.

closed 2024-09-14 by rainmakerz17