OrcaSlicer/OrcaSlicer #6768: OrcaSlicer 2.2.0 beta KlipperScreen jumps away from home and odd error message appears in Klipper

william-orange, 2024-09-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Homebrew similar to an Ender V3, Direct Drive running KLIPPER

### How to reproduce

Print file with adaptive pressure advance enabled and get odd behavior noted.

### Actual results

I am running a test print with the new OrcaSlicer Beta 2.2.0.  So far the print looks great.  I am running Adaptive pressure advance.  If there are any odd defects, I would put it in another bug report.  Here's what I am seeing.

On the KlipperScreen home screen below the name of the print file, Klipper displays the message "RPT Set Ignored".  I have no idea of the meaning of this message.

The second anomoly is that KlipperScreen jumps from the home screen peridically to the macro screen during the first minuites of printing.  Induced by Adaptive Pressure Advance perhaps?

Neither of these oddities occured in prior Orcaslicer versions.

### Expected results

No odd behavior.

### Project file & Debug log uploads

[greek_pot.zip](https://github.com/user-attachments/files/17003872/greek_pot.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bill-orange, 2024-09-14:
My mistake on jumping to the macro screen. The X-stepper cable would periodically hit the touch screen. Looks like the odd error message is the only issue.

william-orange, 2024-09-14:
My error it appears to be a macro issue

closed 2024-09-14 by william-orange