OrcaSlicer/OrcaSlicer #6771: New Language appear in the drop down in preferences but doesn't work

AbdElhalim12, 2024-09-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

all

### How to reproduce

create new language and add the mo file to i18n folder 

### Actual results

I followed the localization wiki page to create a new language that doesn't exist in orcaslicer, then I modified the preferences.cpp file to add add it. The Language now appears in the dropdown in preferences window but when I choose it the orcaslicer needs to be restarted as normal but after restart the language still English. 
I tried to move the mo file to the en folder to test whether it works or not, and it appears correctly. 

### Expected results

new language should take over the UI

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2024-09-17:
You could try raising a draft PR so I could take a better look at what you've got so far.

AbdElhalim12, 2024-09-18:
https://github.com/AbdElhalim12/OrcaSlicer/pull/2
@Noisyfox

AbdElhalim12, 2024-09-18:
https://github.com/SoftFever/OrcaSlicer/pull/6815

Noisyfox, 2024-09-18:
The reason it doesn't work is this:
https://github.com/SoftFever/OrcaSlicer/blob/587e83939777b44c13ca26223fbb3ffe6b2e6863/src/slic3r/GUI/GUI_App.cpp#L5091-L5094

Noisyfox, 2024-09-19:
> but when I replace the english mo file with the arabic one to test it, it worked ![image](https://private-user-images.githubusercontent.com/14279227/368606653-75b4c5cd-25db-4f47-a91d-76425147a054.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY3MDcxOTIsIm5iZiI6MTcyNjcwNjg5MiwicGF0aCI6Ii8xNDI3OTIyNy8zNjg2MDY2NTMtNzViNGM1Y2QtMjVkYi00ZjQ3LWE5MWQtNzY0MjUxNDdhMDU0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE5VDAwNDgxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYwNjg1YzZlMTQwYzBjMzY0N2IxNzMyNjJiMzcxYzc5MzlhZmUzZTJiNGU1OTNhNjFmYWMyNDk4NWVlMGE1NjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RZqfzq_haUvk5lI_2Iz8MlB9nNYwMEelOFxalmkWE8s)

Because in that case the app believes it's English not Arabic, which bypassed this check.

AbdElhalim12, 2024-09-19:
Okaaay, So it there a way to exclude Arabic from that condition, since it's working correctly?

Noisyfox, 2024-09-19:
You can just change the line I mentioned above, (or even delete that check)

AbdElhalim12, 2024-09-21:
Hey, Ive tried couple things but all I got is the whole UI turned Right to left
[image]
[image]

Noisyfox, 2024-09-22:
Ah... I guess that's how wxwidgets handles right-to-left languages.

closed 2024-12-29 by github-actions[bot]