OrcaSlicer/OrcaSlicer #6772: Top layer has a small 1 layer island

wired2000000, 2024-09-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1-2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro 22H2

### Additional system information

CPU: AMD Ryzen 5 2600, RAM: 16GB GRAPHICS CARD: NVIDIA GT710

### Printer

KLIPPER modified Artillery Hornet

### How to reproduce

Open the attached project file.
Slice it.

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[reader helper.zip](https://github.com/user-attachments/files/17005002/reader.helper.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]

That artifact on the top layer should not exist, the stl file does not have it.
[image]
It is not affected by "wall generator settings" neither by "precision" settings
If z axis scale is other than 100% (even 100.1% or 99.9%) then the bug does not appear. I suspect some floating point arithmetics shenanigans.
[image]

discip, 2024-09-16:
@wired2000000
Unfortunately your model is corrupted.

Right click your model and choose **`Fix Model`**. 😊

discip, 2024-09-16:
@wired2000000
This is no bug!
Unfortunately your model is corrupted.

Right click your model and choose **`Fix Model`**. 😊

discip, 2024-09-16:
@wired2000000
This is not a bug in Orca!
Unfortunately your model is corrupted.

Right click your model and choose **`Fix Model`**. 😊

discip, 2024-09-16:
@wired2000000
This is not a bug in Orca!
Unfortunately your model is corrupted.

Right click your model and choose **`Fix Model`**. 😊

[image]

closed 2024-12-24 by github-actions[bot]