OrcaSlicer/OrcaSlicer #6773: Add Z coordinate when come back to purge tower

CorvoConV, 2024-09-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Good afternoon.
Reviewing the tool change code, after sending, for example, T1, the gcode back to the purge tower does not take into account the Z coordinate. This is a problem because in the tool change macros the head must be maintained at the height of the printing layer and when the head goes to the purge tower it can hit it. I've had this problem with older versions of PrusaSlicer but simply adding the Z coordinate in the gcode back to the purge tower fixed it.
This is an example of what I mean:
ORCASLICER
image

PRUSASLICER
image

Can you improve this please?
Thanks a lot.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Just add the layer Z coordinate when come back to purge tower

AccelGuide, 2024-09-15:
you can specify the name and type of printer,
also a project upload for testing?

CorvoConV, 2024-09-15:
Hi.
The printer its a custom IDEX i build, but the printer does not matter. Its a slicer problem.
In older versions of PrusaSlicer that makes just like now OrcaSlicer, with out the Z coordinate when they come back to the purge tower but in the new version they fix that. 
Here is a simple print i do just to calibrate tool offset of my IDEX printer:
[image]
Its a simple cube that allows me to measure in X and Y with a caliper and know the offset to fix it.
I put real GCODE from both cases for the tool change part in ORCASLICER and in PRUSASLICER.

ORCASLICER:
[image]

PRUSASLICER:
[image]

Some times ORCASLICER does not put the Z coordinate when come back to the purge tower.
And other issue its that the first pruge in the tower from a new tool its made in the last layer height:
[image]

I attach both gcode files.
Thanks for your help.
[CUBES.zip](https://github.com/user-attachments/files/17005537/CUBES.zip)

AccelGuide, 2024-09-15:
I would like to help, but without the proper .3mf file, I can't be certain. It seems like it's just a filament change G-code issue. Please go to the filament G-code section in PrusaSlicer and copy its content. There will likely be a line like "layer_z" that's missing, which causes the G-code to fail writing it into the final G-code.
[image]

Could you please provide the content of the PrusaSlicer filament change G-code, and maybe also the prime tower G-code? Also, try to import a section/line with a semicolon "; Test Z" into Orca Slicer. Can you do that, or am I on the wrong path?

CorvoConV, 2024-09-15:
Hi.
It is for sure a slicer problem. PrusaSlicer fix that in one of the latest updates and they put in the bug fix text as a issue:
[image]

My layer change gcode its just exact like yours in prusaslicer and orcaslicer. I have the factory gcode configuration for all, layer, tool change, etc. Only make changes on start and end gcode. I try to make changes with the semicolon and diferent gcodes but does not works. Just OrcaSlicer not emit the z position when come back to purge tower.
I attach the 3mf file.
Thanks.
[CUBE_ORCA.zip](https://github.com/user-attachments/files/17005895/CUBE_ORCA.zip)

AccelGuide, 2024-09-15:
Thanks for the Info:
Link:
https://github.com/prusa3d/PrusaSlicer/releases?page=2

https://github.com/prusa3d/PrusaSlicer/commit/fa23ab82853f6b033b1ceea01d4932a5aae1b013

###
So the Solution is to make the same cange like:

Make sure that after wipe tower toolchange the G1 Z is always emitted.

Even if the G1 Z would be redundant, it is still emitted.
Fixes: https://github.com/prusa3d/PrusaSlicer/issues/12361
###

I will try to get in contackt witch : [lukasmatena](https://github.com/prusa3d/PrusaSlicer/commits?author=lukasmatena)

CorvoConV, 2024-09-15:
Thanks a lot.
In PrusaSlicer was the solution to my printing problems with my IDEX printer.
This is just the last step to move my IDEX profile to OrcaSlicer ;)

AccelGuide, 2024-09-22:
@lukasmatena Hi, 
I Would Like to try my first Change Here on This issue/ Enhancement and would like to question a Code Review, wenn I'm finished?

I have found out that you have made Last time the Change in Prusa Slicer so it Word be Nice if you Could then Help me Check the Code Change and Review, and some Information about "How to" on the Orcas Slicer Projekt.

igiannakas, 2024-09-27:
> Thanks for the Info: Link: https://github.com/prusa3d/PrusaSlicer/releases?page=2
> 
> [prusa3d/PrusaSlicer@fa23ab8](https://github.com/prusa3d/PrusaSlicer/commit/fa23ab82853f6b033b1ceea01d4932a5aae1b013)
> 
> So the Solution is to make the same cange like:
> 
> Make sure that after wipe tower toolchange the G1 Z is always emitted.
> 
> Even if the G1 Z would be redundant, it is still emitted. Fixes: [prusa3d/PrusaSlicer#12361](https://github.com/prusa3d/PrusaSlicer/issues/12361)
> 
> I will try to get in contackt witch : [lukasmatena](https://github.com/prusa3d/PrusaSlicer/commits?author=lukasmatena)

This feature was already in 2.2 beta. But it causes this issue - last layer nozzle crash when a single filament change happens on the previous layer.
https://github.com/SoftFever/OrcaSlicer/pull/6894#issuecomment-2379439695

53Aries, 2024-10-13:
I don't know if this helps you, but here's how I handle Z moves during toolchange. 
I use macro Lift_and_Go_to_Purge before the T{next_extruder} command. This causes the toolhead to Z hop the amount specified by {% set z_lift = params.Z_LIFT|default(0)|float %} which is defined in the Change filament G-code section as Lift_and_Go_to_Purge Z_LIFT=1.0. After the toolchange is complete I use macro Move_Back_to_Print which gets it's positions from Change filament G-code section Move_Back_to_Print X={x_after_toolchange} Y={y_after_toolchange} Z={z_after_toolchange}. 
The actions look like this: Nozzle moves to prime tower, z hop, toolhead does what it needs to do for toolchange(moving to purge, wiping, etc) Once toolchange is complete the toolhead moves back to the prime tower at the correct Z height to begin printing. 

[gcode_macro Lift_and_Go_to_Purge]
# params:
#   Z_LIFT  distance to lift the z axis before moving to the purge location
gcode:
    {% set z_lift = params.Z_LIFT|default(0)|float %}
    {% set z_max = printer.toolhead.axis_maximum.z - printer.gcode_move.homing_origin.z %}
    {% set z = printer.gcode_move.gcode_position.z %}
    {% set z_park = [z + z_lift, z_max]|min %}

    # lift z
    G90                     # absolute positioning
    G1 Z{z_park}            # move to z position

    # move to purge location
    Go_To_Purge_Location

[gcode_macro Move_Back_to_Print]
# params:
#   X   x position to move to
#   Y   y position to move to
#   Z   z position to move to
gcode:
    {% set x = params.X|float %}
    {% set y = params.Y|float %}
    {% set z = params.Z|float %}
    {% set st = printer["gcode_macro TR_Variables"].travel_speed * 60 %}
    
    # move back to print
    G90                     # absolute positioning
    G1 X{x} Y{y} F{st}      # restore original x and y position
    G1 Z{z}                 # restore original z position

{if layer_num >=0}
Lift_and_Go_to_Purge Z_LIFT=1.0
M106 S0
T{next_extruder}
PURGE FV={flush_length} FE={new_filament_e_feedrate}
M106 S0
Wipe_Nozzle
Move_Back_to_Print X={x_after_toolchange} Y={y_after_toolchange} Z={z_after_toolchange}
{endif}

closed 2025-01-20 by github-actions[bot]

JuanR3D, 2025-02-03:
Hey @53Aries 

I use something similar in order to handle toolchanges.
I send the toolhead directly in my t0 and t1 macros to the next print position, But for some reason, x_after_toolchange and y_after_toolchange are now working in my orcaslicer, and i need to postprocess the gcode with a scrip i wrote in order to get those values correctly.

Could you slice a piece and see if those x_after_toolchange and y_after_toolchange are working propperly in your slicer?

I send you a photo of what is happening in my orcaslicer:

[image]

JuanR3D, 2025-02-03:
as you can see, x_after_toolchange in my case is 161.139, but the G1 command shows that this value is 330.056

JuanR3D, 2025-09-21:
> Hey all,
> 
> Thank you for the feedback. I just want to double-check: is the issue that your IDEX printer firmware changes the Z position during tool changes and does not restore the Z height afterward? Are you expecting the slicer software to force-restore the correct Z position? Is this correct?
> 
> If so, please try this PR [#10803](https://github.com/SoftFever/OrcaSlicer/pull/10803) and let me know if it resolves the problem. If not, please leave a comment and let me know. Thank you.
> 
> Thanks -SoftFever

In my case yes. I do a z-hop when comanding a toolchange but if z after toolchange coordinate is not defined, it will continue printing mid air. 

I'll try your PR and give feedback as soon as posible.

Anyways, in my case, I solved it by giving the X Y Z coordinates via a gcode postprocessor. For example, now my T0 and T1 macros are like: T0 X955 Y400 Z600

JuanR3D, 2025-09-21:
> Hi. The printer its a custom IDEX i build, but the printer does not matter. Its a slicer problem. In older versions of PrusaSlicer that makes just like now OrcaSlicer, with out the Z coordinate when they come back to the purge tower but in the new version they fix that. Here is a simple print i do just to calibrate tool offset of my IDEX printer: [image] Its a simple cube that allows me to measure in X and Y with a caliper and know the offset to fix it. I put real GCODE from both cases for the tool change part in ORCASLICER and in PRUSASLICER.
> 
> ORCASLICER: [image]
> 
> PRUSASLICER: [image]
> 
> Some times ORCASLICER does not put the Z coordinate when come back to the purge tower. And other issue its that the first purge in the tower from a new tool its made in the last layer height: [image]
> 
> I attach both gcode files. Thanks for your help. [CUBES.zip](https://github.com/user-attachments/files/17005537/CUBES.zip)

"By the way, Corvo, I don’t know if you’ve already found it, but it’s easier to calibrate XY offsets with something like this:"
https://www.printables.com/model/129617-offset-xy-dual-extruder-idex-calibration