OrcaSlicer/OrcaSlicer #6774: Paint brim

Dani7710, 2024-09-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

the problem is that when you print multiple things, some objects need brim, other objects don't

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

having the ability to draw the brims, as well as the supports and seams would be a great thing when you have multiple items on the plate. I leave you a modified screenshot with the idea. I hope the drawing is understandable.

[image]

Arakon, 2024-09-15:
You can already set brim on/off by object.
[image]

Dani7710, 2024-09-17:
> You can already set brim on/off by object. [...]

however there are large objects that need to have the brim only on protruding corners, I think the ability to draw the brim only where you want is not bad

execsumo, 2024-11-09:
Also have this problem frequently; most surfaces of the object I'm printing don't need a brim, but a small part does. Adding a brim to the whole object leaves unnecessary, jagged edges from removing brim.

yazz007, 2025-02-14:
Brim Painter is a must. It’s quite difficult to place it in the right spot using Mouse Ears, so +1 from me.

closed 2025-02-18 by SoftFever

slicemans, 2025-10-25:
what about brim in new 2.3.1 version

commit df902d13 (2025-02-18): Port Paint Brim Gizmo from BBS (#8433)

This PR ported the Brim Ear gizmo from BambuStudio. Thanks BambuLab!

For compatibility reason, the existing auto-detected brim ears is kept.
To use this new brim ear gizmo, you will have to change the brim style
to "Painted".

[image]

Fix #6774