OrcaSlicer/OrcaSlicer #6777: Add a possibility to do not unload the filament after the print for Bambu AMS

aremishevsky, 2024-09-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The filament is always unloaded after the print is completed if it is using the AMS. 
This might be useless if you print a lot of the same parts.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Add a checkbox before send to print to do not unload the filament after the print.

Arakon, 2024-09-15:
That's part of the machine end print gcode, not something Orca just does by itself. You can edit that to your liking.
i.e. 
[image]

Ic33s, 2024-09-15:
As said, it's part of the end gcode. You can create a second machine profile with this change added. (I've done to improve end position and startcode for improving nozzle cleanup). So you can easily switch between them.

Btw. I experienced some filaments to break if left in the ptfe "system" for to long. So if you forgot to change the setting, it might be more work to clean the ptfe system from filament than it really helps. So you should check that after the last part your filament is unloaded.

closed 2024-12-15 by aremishevsky