OrcaSlicer/OrcaSlicer #6781: Improve shrinkage support for multimaterial/multicolor prints using filament with different shrinkage values.

Ic33s, 2024-09-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I always need to edit shrinkage for several filaments on multimaterial/multicolor prints, so they match a specific one (mostly the one mostly used in the print object. I know I would also be able to rescale manually but shrinkage is given a different way than I need to easy rescale.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would love to be able to select a filament the shrinkage is used from for rescaling a print object. This should be automaticly set to the filament most used in a print object, but I can change it before slicing if I want to. The setting should be changebale by print object. But a global one would already help a lot.

Thank you very much.

### Describe alternatives you've considered

1. Manually: Changing (but not accidently saving to the filament profile!!!) all filament profile I use in a print to match the one I want
2. Manually: Rescale object manually (unfortunatly the shrinkage is given in an other way than I need for rescaling)

Ic33s, 2024-10-30:
Because I became aware of this today:
In addition please just ignore shrinkage informations coming from material only used for support base / support interface all together and only put the materials actually used for the object itself into consideration for the shrinkage calculation.

Ic33s, 2024-12-15:
Couldn't imagine nobody else has this problems with multimaterial prints. Hope there might be a solution. Thanks :)

lromeraj, 2025-01-13:
I've encountered a similar situation to what you described. I typically pay close attention to part dimensions, and sometimes I add signatures in different colors to the parts. In my case, I was working with a combination of PLA and PLA-CF, which have slight differences in shrink factors. This caused the slicer to flag issues due to the mismatch in shrink factors.  

Another challenge arises when mixing material types for supports. For instance, when stacking multiple parts on top of each other, I sometimes design a specific part solely for support. In such scenarios, the slicer automatically disables shrink compensation, which isn't ideal.  

When dealing with multiple parts on the plate, as shown below:  
[image]  

It would be helpful if the slicer could apply the shrink factor of the material used for the topmost part. This approach could account for the rest of the parts grouped under the name `cd_plate (2)` in this example.

skl111, 2025-03-02:
Yes Please Add this or a way to bypass it as I am using PLA and PETG with support materials and I have to manually scale the models due to the support filament having 100 and PLA/PETG having around 99.6-99.3 when measuring 100mm shrinkage.

I would think this would be done before slicing so it should not affect the the sliced data when using support materials

ergonomicmike-2, 2026-10-15:
Instead of creating a new Request for Enhancement related to ignoring Filament Shrinkage, I'll add on here.

As a first Pass toward implementing this Enhancement, could Orca check to see if only one tool head is being used for a print? And if so, then NOT ignore Filament Shrinkage?

I have a new SnapMaker U1. I have dedicated T4 for TPU. (SF nozzle, and I've pulled the PTFE Tube out of the feeder and use manual loading for T4.) I've assigned a Filament Shrinkage value to my Process for TPU SF. 

But when I print a model using only T4, Orca warns that Filament Shrinkage will not be used because "the used filaments do not match." 

But there are no other used filaments. 

It appears that Orca is assuming that because any other Toolhead than T1 is being used, then other filaments are being used. But that's not necessarily true.

So a Sanity Check, to check that only one Toolhead is in use, might be a first step in implementing this overarching Request. 

Or the first step in implementing [a similar Request (for a two material print where the 2nd material is only support material.?(https://github.com/OrcaSlicer/OrcaSlicer/issues/8756)