OrcaSlicer/OrcaSlicer #6786: New Adaptive Pressure Advance Calibration Style

ruredi87, 2024-09-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It takes very long to calibrate for adaptive pressure advance. I mean We have to slice seperately and print for every different speed/acceleration value.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

We should be able to input for speed and acceleration steps, so advanced pa calibration could be done at one step.

### Describe alternatives you've considered

.

Ic33s, 2024-09-16:
I would like to add, it would be greate if speed and acceleration values are calculated automaticly from the print/filament profiles, or there might be a useable default set (maybe per filament type). I'm still not sure what PA calibrations I need to do for using the feature and how to set them up correctly.

Andreas-strg, 2024-09-22:
I also think also we need a new calibration tool for the adaptive pressure advance that makes all at once
The manual handwork are to mutch...

ruredi87, 2024-09-22:
Definetly. We should enter acceleration/speed values and it should create patterns which could even cover all the print bed, like bunch of PA calibration patterns.

FlorianSchreiner1, 2024-11-10:
I'm just going through the calibration process and this would definitely be a worthwhile feature to add. The time required for the calibration isn't the bad part but rather the fact that one has to start a new print every ~15 min with manually set up tests.

DearTanker, 2024-11-19:
Hopefully this feature will be released soon, which will save a lot of calibration time.

closed 2025-02-25 by github-actions[bot]