OrcaSlicer/OrcaSlicer #6787: [toolchanger] Allow movement to wipe tower before tool change.

N3MI-DG, 2024-09-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It would be great to have the wipe tower X/Y positions available to use in filament change gcode so that we can move to the wipe tower before changing filament. Stealthchanger returns to the last known Z position after changing tools, but has no knowledge of the position of the wipe tower. For us to move the next tool to the wipe tower after toolchange we need to move to it before changing. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add menu option to move to wipe tower before change.

### Describe alternatives you've considered

1. Expose wipe_tower_x and wipe_tower_y for use in filament change gcode.

### Additional context

wipe_tower_x and wipe_tower_y are available as variables in the tool change gcode section of PrusaSlicer to achieve this need.

wwmotorsport, 2024-10-04:
I agree, this would be a huge improvement. I'm currently running several IDEX printers, and the ooze from the idle nozzle gets dragged into the print after the tool change before moving to the wipe tower. Not great...

npoirey, 2025-01-04:
I would like that as well

tomas-a-sanchez, 2025-01-16:
were this addressed?

tylerhartwig, 2025-01-30:
Is this not possible in OrcaSlicer?

I was just looking at this, I'm able to access `wipe_tower_x` and `wipe_tower_y` in the tool change section of my printer profile.

only difference is that in Orca it's an array rather than a variable, I think it points to each corner. so you should be able to do a G1 to `wipe_tower_x[0]` and `wipe_tower_y[0]` -- in theory you could also calculate the center by utilizing the second coordinates in the array as well.

tomas-a-sanchez, 2025-01-31:
I just add a line while they look for make it that way by default. 
I add to the line my code: G1 Z{layer_z+8} X30 Y240 F4500 so in that way, anytime there is a tool change it does the change at Y240, were most of the time I put my wipe tower.

tylerhartwig, 2025-01-31:
Ah got it, you just want it to be available in the UI.

You could try `G1 Z{layer_z+8} X{wipe_tower_x[0]} Y{wipe_tower_y[0]} F4500` if you don't want to have to put the wipe tower in the same place every time

tomas-a-sanchez, 2025-01-31:
Ohhh GOD!!!!! That's amazing my friend! thanks a lot! I am not that into G-codes that's for pros. I will try this and provide feedback. Also will be great if that could come by default!

wwmotorsport, 2025-02-10:
Oh man, this is incredible! Thanks for that simple piece of g-code. It works perfectly, and seems like this should be the default behavior of the slicer.

tylerhartwig, 2025-02-10:
I think different machines and setup may need different settings when it comes to tool change position, likely why it's not default for the slicer, but maybe should be default for specific printers