OrcaSlicer/OrcaSlicer #6789: Inner/Outer/Inner wall order prints some outer perimeters last, causing poor dimensional accuracy

Snelso91, 2024-09-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

Open attached 3mf project and slice.

### Actual results

Look at layers 2-17.
On these layers, the 4 circular holes and the central large rectangular hole all have their outer walls (orange) printed last, after all the inner walls have already been printed. This causes all 5 of these holes to be printed ~0.2mm undersized (diameter) due to their outer walls being pushed out by the inner walls, just like with Inner/Outer wall order. This issue occurs with both Arachne and Classic modes.
[image]

If you change the wall order from Inner/Outer/Inner to Outer/Inner and re-slice, the problem goes away - the outer walls are correctly printed first and the dimensions come out correct.
[image]

### Expected results

What I expect to happen when using Inner/Outer/Inner (IOI) wall ordering, is that outer walls should NEVER be printed after any directly adjacent inner walls, in order to maintain good dimensional accuracy and surface finish just like with Outer/Inner (OI) wall ordering.
This expectation is reinforced by the wall order tooltip that describes IOI as being recommended over OI in most cases because it is supposed to have the benefits of better dimensional accuracy and surface finish like OI, but with the additional benefit of hiding seams better.

If for some reason the wall ordering algorithm cannot produce a result that does the above, then there should be an automatic fallback to use OI wall ordering on that particular layer/area of the layer.

If for some reason falling back to Inner/Outer (IO) is preferred, then I suppose there could be a user-selectable fallback option to choose between either OI or IO.
However in my opinion, falling back to IO makes little sense in most cases since it seems like IOI is supposed to be like a better/improved OI, and by falling back to IO you would lose out on the benefits that you would expect from OI or IOI.

### Project file & Debug log uploads

[Bambu Lab P1S Camera Bracket.zip](https://github.com/user-attachments/files/17008616/Bambu.Lab.P1S.Camera.Bracket.zip)
[recent logs.zip](https://github.com/user-attachments/files/17008617/recent.logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DonutCables, 2024-10-10:
I've also run into this same issue. In my case, on layer 49 of my file the outermost wall of the part is printed last compared to the inner walls on most of the part. It seems to happen when there's no single inner wall around the entire part due to other geometry in the part causing smaller wall sections. I've attached the project 3mf and gcode I'm seeing this happen with.
[example_file.zip](https://github.com/user-attachments/files/17320813/example_file.zip)

justr0st, 2024-10-16:
Was about to post the same issue, but it exisists already. The internal hole dimensions are inconsistent when compared to the external ones because of this.

closed 2025-01-23 by github-actions[bot]

Snelso91, 2025-01-31:
I can confirm this issue is still present in OrcaSlicer 2.2.0, so the issue has not been fixed.

igiannakas, 2025-02-01:
> I've also run into this same issue. In my case, on layer 49 of my file the outermost wall of the part is printed last compared to the inner walls on most of the part. It seems to happen when there's [...]

I believe this is fixed in the latest nightly releases.

[image]

Also for @Snelso91 again this looks correct in the latest nightly.

[image]

MartynasJonkus, 2025-05-05:
Still not working correctly for me on either 2.3.0 or 2.3.1 nightly. Problem arises when outer and inner perimeters are too close to each other.  Inner perimeter's outer wall is being printed as the last wall here.
[image]

igiannakas, 2025-05-05:
In the top right corner of your model the first inner perimeter is also neighbouring the inner contour external wall.

Hence it’s prioritising external wall appearance there and printing the inner wall last. 

There are so many variations to this that not all cases can be perfectly caught. At present this gives good enough results for the vast majority of cases - where external wall finish is desired the outcome is always consistent. 

Inner contours can be de prioritised where the algorithm cannot deterministicaly make a decision on wall ordering.

If you’re having accuracy issues enable precise outer wall. This should help regain accuracy in the inner contours

go-hacks, 2025-08-15:
This problem still exists in version 2.3.1-dev. Inner/outer/inner is printing the same as Inner/outer. With 3 walls it does the inside then the middle then the outside. It should obviously do the inside, the outside, and then the middle.

igiannakas, 2025-08-15:
Post a project file please