OrcaSlicer/OrcaSlicer #6797: Ability to split a model by color

JAHMCR, 2024-09-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I often find myself in the situation of wanting to print a multi-color model, to optimize printing time and material used, I like to split the models into parts so I can print them separately in different colors, but sometimes the models are too complex to be split in Orca Slicer (I have no experience with modeling software).

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to make a suggestion to use the existing painting function to split models by color.

### Additional context

For example, I would like to split a snowman into: body, arms, eyes, nose, buttons and hat so I can print them separately and then glue them together.

delovelady, 2024-10-16:
Desparately needed function.  This is an example model (and there are thousands) that could benefit from this capability: https://www.printables.com/model/705941-power-of-prayer

closed 2025-01-23 by github-actions[bot]

cybirdcosmo, 2025-09-21:
I've converted .3mf to .zip and started analyzing the color specifications of the colors and how they're resolved in the mesh in the Bambu slicer. Im pretty sure the color state of each object is in the Zip_or_3MF_file\Metadata\model_settings.config under   `<metadata key="extruder" value="1"/>
`
And the mesh is at 

Zip_or_3MF\3D\Objects\object_##.model
## - represents an arbitrary number

They are unfortunately not clearly specified for easy parsing 

```
<triangle v1="48" v2="46" v3="50" paint_color="001C21C1C21C1C21C1C21C1C21C1C21C1C1C3"/>
     <triangle v1="46" v2="51" v3="50" paint_color="1C"/>
     <triangle v1="52" v2="48" v3="50" paint_color="1C01C1C21C1C21C1C21C1C21C1C61C01C1C21C1C21C1C21C1C61C01C1C21C1C21C1C61C01C1C21C1C61C01C1C61C00AAAAAA"/>
     <triangle v1="52" v2="53" v3="48"/>
```