OrcaSlicer/OrcaSlicer #6798: Variable Layer Height lowest layers not selectable

ZedOud, 2024-09-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

macOS

### OS Version

Sonoma 14.6.1

### Additional system information

CPU: Intel

### Printer

Elegoo Neptune 4 Max

### How to reproduce

1. Add a primitive.
2. Open the Variable Layer Height menu.
3. Attempt to manually paint the bottom 5% on the layer representation side bar (with left-click or right-click).

### Actual results

It cannot be modified (as shown in the attached screenshot. 
Maybe because the pop-up UI element stating the height gets in the way?

### Expected results

As you can see in the screenshot, the layers which are grey should be able to be modified.

### Project file & Debug log uploads

[debug_Mon_Sep_16_15_20_00_63899.log.0.zip](https://github.com/user-attachments/files/17021005/debug_Mon_Sep_16_15_20_00_63899.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Probably related to #6722
[image]

discip, 2024-09-17:
@ZedOud
That is no bug, it is the value set in **`Quality`** => **`First layer height`**. 😊

ZedOud, 2024-10-16:
> @ZedOud That is no bug, it is the value set in **`Quality`** => **`First layer height`**. 😊

That makes sense.

closed 2024-10-16 by ZedOud