OrcaSlicer/OrcaSlicer #6800: Cannot Run AppImage on Linux Missing Dependency

DeathTrooperA01, 2024-09-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Kubuntu 24.04

### Printer

NA

### How to reproduce

Trying to Lanuch Program in Terminal with 

`./OrcaSlicer_Linux_V2.1.1.AppImage`

### Actual results

Get the following error from Terminal

_/tmp/.mount_OrcaSle4u5yA/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory_

### Expected results

Orca should launch

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This appears to be and issue in Bambu Studio as well 
https://github.com/bambulab/BambuStudio/issues/3973

purpl3f0xsecur1ty, 2024-09-18:
+1 on this. This has been an issue for me since upgrading from Kubuntu 23.10 to 24.04. The missing library no longer seems to be in the repos, so I'm guessing it's considered obsolete. I guess we will have to wait until the AppImage is updated to require a newer version of `libwebkit2gtk`.

Constant-Leon, 2024-09-20:
+1 here as well. MX-Linux.

DeathTrooperA01, 2024-09-20:
> I had an issue as well but solved it using the advice in this thread.
> 
> #3160

this did not fix my problem

Roberies, 2024-09-22:
I was able to get around this on a Debian system with: `sudo apt install libwebkit2gtk-4.0-37`

luigicampanaro, 2024-09-24:
This worked for me:

`./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-beta2.AppImage --appimage-extract 
`
`cd squashfs-root 
`
`./AppRun`

JaCKoP619, 2024-09-26:
i've had the same problem from the start that i've tried using orca on ubuntu. I did manage to get a workaround and integrate this package somehow but i don't really remember how now.

f1nch87, 2024-09-27:
> This worked for me:
> 
> `./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-beta2.AppImage --appimage-extract ` `cd squashfs-root ` `./AppRun`

Did not work for me

tomoinn, 2024-09-30:
You need to set WEBKIT_DISABLE_DMABUF_RENDERER=1 in your environment and then you should get the home and project UI pages working. It'll work fine from the app image as long as you use the right one for your version of ubuntu (or ubuntu-like, I'm using Mint for example)

purpl3f0xsecur1ty, 2024-09-30:
> You need to set WEBKIT_DISABLE_DMABUF_RENDERER=1 in your environment and then you should get the home and project UI pages working. It'll work fine from the app image as long as you use the right one for your version of ubuntu (or ubuntu-like, I'm using Mint for example)

Nice! Now my Orca Slicer GUI works perfectly, I can see the home page and access additional printer options.
For anyone else with the same issue, I made this environment variable persistent by adding it to `/etc/environment` and then logged off and back on.

jamwaffles, 2024-10-11:
This is a horrible way to fix things, but I symlinked the 4.1 version of the library to the 4.0 the AppImage is looking for:

```bash
sudo ln -s /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0.5.7 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18
```

This is on Linux Mint 22. The app starts up but I haven't tested it any further as of writing this comment. YMMV.

vkukk, 2024-11-10:
$ ./Bambu_Studio_linux_fedora-v01.10.00.81.AppImage
/tmp/.mount_Bambu_8U27Gn/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

On Fedora 41.

sevenrats, 2024-12-13:
So, there are two releases, "new linux" and "old linux"  22/bookworm appears to be "old", whereas noble/trixie appear to be "new." Maybe this issue is as simple as naming releases in a more conventional way. "old and new" is kinda just asking for this problem.

rubienr, 2025-02-01:
If upgrading from 24.04 to 24.10 is possible, then it would fix the issue. I could build successfully on Ubuntu 24.10.

hth

closed 2025-05-10 by github-actions[bot]