OrcaSlicer/OrcaSlicer #6806: Bridge counterbore holes (partial) applied in all holes, not only small

KubaMiszcz, 2024-09-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

Bambulab X1C

### How to reproduce

model with not so small hole eg 30x40mm (not necessary circle), and check: Bridge counterbore holes, set: partial

### Actual results

it is treated like small hole, and cover with two layer, which is expected in small holes
actual:
[image]

### Expected results

add some trigger, eg by hole area, overall dimension X/Y etc
here is with this option off:
[image]

### Project file & Debug log uploads

[test.zip](https://github.com/user-attachments/files/17029823/test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

discip, 2024-09-17:
@KubaMiszcz
Is there a particular reason why you don't just turn it over? 🤔

KubaMiszcz, 2024-09-18:
because I need to remember to check for it, and i think intention of this feature is to target small holes like for scres/bolts/pins

discip, 2024-09-18:
Sorry, I'm not getting your point.
So ... if I got you right, you are intentionally searching for cases where this feature starts to fail?

KubaMiszcz, 2024-09-18:
nope, this feature is very useful and i have it always on,
but I've had a problem with this hole from picture where i need to have smooth edge, and realize the reason

in bigger holes it is useless, bridges are too long and you need support anyway, and it is no matter if you have it on or not

closed 2025-03-27 by github-actions[bot]